google/brax #142: <canvas> for HTML brax-viewer ignores specifying "height"

ymd-h, 2022-01-03:
Thank you for providing a great project!

I noticed HTML viewer (`brax@v0.0.10/js/viewer.js`) ignored specified height and became as big as possible.
(Moreover, sometimes the viewer overwraps other outputs.)

## Reproducible Code on Google Colab
```python
!pip install brax

from IPython.display import HTML
from brax.io import html
from brax import envs
import brax.jumpy as jp

env = envs.create("ant")

state = env.reset(jp.random_prngkey(0))

display("This message triggers size increase.")
display(HTML(html.render(env.sys, [state.qp], height=10)))
display("This message is overwrapped by viewer.")

html.save_html("test.html", env.sys, [state.qp]) # Download "test.html" and view with browser.
```

## Tested
- Brax 0.0.10
- Google Chrome
- macOS Big Sur 11.6
- Environment
  - Google Colab
  - Local html file (written by `brax.io.html.save_html`)

## Backgound
`brax.io.html.render` function takes `height` parameter and creates `<div>` element with fixed height.
I expected the output viewer had fixed height, however, it is not true.

https://github.com/google/brax/blob/661364ee04f2e31017bea40e02eee6968bde5af2/brax/io/html.py#L40
https://github.com/google/brax/blob/661364ee04f2e31017bea40e02eee6968bde5af2/brax/io/html.py#L64-L68

## (Possible) Cause
At some events, the viewer is resized without specifying explicit height (and width). If height parameter (`h`) is `undefined`,  `window.innerHeight` is used instead of the parent `<div#brax-viewer>`'s height.

https://github.com/google/brax/blob/661364ee04f2e31017bea40e02eee6968bde5af2/js/viewer.js#L168-L170
https://github.com/google/brax/blob/661364ee04f2e31017bea40e02eee6968bde5af2/js/viewer.js#L183-L189

## (Candidate) Fix
I think there are (at leaset) two candidate fixes.

### Candidate 1
The first fix is replacing the default fallback value of `h` in `setSize()` by `this.domElement.clientHeight`.
```python
   if (h === undefined) { 
     h = this.domElement.clientHeight; 
   } 
```
### Candidate 2
The other fix is using `resizeCanvasToDisplaySize()` as `ResizeObserver` does.

https://github.com/google/brax/blob/661364ee04f2e31017bea40e02eee6968bde5af2/js/viewer.js#L172-L173
https://github.com/google/brax/blob/661364ee04f2e31017bea40e02eee6968bde5af2/js/viewer.js#L201-L206

erikfrey, 2022-01-05:
Hi @ymd-h - thanks for pointing this out!  It looks like Candidate 1 most directly addresses the bug, that looks like a good solution to me.  Would you like to open a PR?

ymd-h, 2022-01-06:
@erikfrey 
Thank you for your reply!

> Would you like to open a PR?

Yes, I made a PR. (#146)

I couldn't find any test codes for JavaScript, however, at least I checked with following procedure;
1. I manually bundled all the JavaScript codes and embedded into html template string.
1. I ran it on Google Colab and checked it worked as I expected.

erikfrey, 2022-01-07:
OK, this is merged.  Thanks!

closed 2022-01-07 by erikfrey