google/brax #147: Unsupported kUInt bits 32

conglu1997, 2022-01-07:
I get an error "Unsupported kUInt bits 32" when using the environments "grasp", "fetch", "ur5e"  with the PyTorch version of brax provided in the notebook: [https://colab.research.google.com/github/google/brax/blob/main/notebooks/training_torch.ipynb](https://colab.research.google.com/github/google/brax/blob/main/notebooks/training_torch.ipynb)

```
/usr/local/lib/python3.7/dist-packages/torch/utils/dlpack.py in from_dlpack(ext_tensor)
     65         # Old versions just call the converter
     66         dlpack = ext_tensor
---> 67     return _from_dlpack(dlpack)

RuntimeError: Unsupported kUInt bits 32
```

To reproduce this issue simply replace the env name in the train function:

```
def train(
    env_name: str = 'fetch',
```

erikfrey, 2022-01-13:
Thanks for catching this!  It should be addressed in 1baf25d5a713bd5dbc8588a004a5754723626bd0 - please let us know if you still have problems.

closed 2022-02-09 by conglu1997

ishitamed19, 2022-02-24:
@erikfrey Hi, I am facing the same issue with "Grasp" and "Fetch" on my system (when I copy the code from the colab). Here is the error:
```
Traceback (most recent call last):
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/private/home/ishitamed/projects/multiagent_brax/train.py", line 359, in <module>
    train(env_name=args.env_name, num_envs=args.num_envs, episode_length=args.episode_length,
  File "/private/home/ishitamed/projects/multiagent_brax/train.py", line 75, in train
    env.step(action)
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/site-packages/brax/envs/to_torch.py", line 63, in step
    info = self.info(info)
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/site-packages/brax/envs/to_torch.py", line 51, in info
    return torch.jax_to_torch(info, device=self.device)
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/functools.py", line 877, in wrapper
    return dispatch(args[0].__class__)(*args, **kw)
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/site-packages/brax/io/torch.py", line 96, in _jax_dict_to_torch
    **{k: jax_to_torch(v, device=device) for k, v in value.items()})  # type: ignore
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/site-packages/brax/io/torch.py", line 96, in <dictcomp>
    **{k: jax_to_torch(v, device=device) for k, v in value.items()})  # type: ignore
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/functools.py", line 877, in wrapper
    return dispatch(args[0].__class__)(*args, **kw)
  File "/private/home/ishitamed/.conda/envs/brax/lib/python3.9/site-packages/brax/io/torch.py", line 84, in _devicearray_to_tensor
    tensor = torch_dlpack.from_dlpack(dpack)
RuntimeError: Unsupported kUInt bits 32
```

Dependencies installed are:
```
brax                      0.0.10
python                    3.9.7
jax                       0.3.0 
jaxlib                    0.3.0+cuda11.cudnn805
pytorch                   1.9.0           py3.9_cuda11.1_cudnn8.0.5_0
```

I am using cuda11.1 with cudann v8.1.1.33-cuda.11.0 on my system