google/brax #149: Tendon simulation in Brax

pete-j-turnbull, 2022-01-10:
I'm looking to simulate a tendon within Brax so that I can model a dexterous hand and train an RL agent to move it naturally.

This works in Mujoco right now by treating the muscle connected to the tendon as a 0 length actuator whose force depends on how contracted the 'muscle' is. Ie. the tendon actually changes length based on the force applied to it.

I'd like to have a go at implementing this and potentially submitted a PR - does anyone have any tips on how this might be implemented best within the Brax framework.

Thanks!

cdfreeman-google, 2022-01-10:
Hello!

I'd expect this to look very similar to our angle-actuator defined here: https://github.com/google/brax/blob/main/brax/physics/actuators.py#L72

Do you have a particular form for the biodynamics you're trying to model beyond modulating the actuator spring constant?

pete-j-turnbull, 2022-01-11:
Heya

Thanks for the tip on the angle actuator - it seems like it could be very useful. The scenario I want to be able to model is a tendon attached to a contracting muscle that goes over a number of pulleys (cylinders that can rotate) and finally connects to a bone (capsule). Then when the muscle contracts it has the effect of shortening the tendon and generating a force which is translated through the pulleys to the bone (capsule) at the end.

I think the various forces involved aren't actually all that complex - its just that there's quite a lot going on all at once in order to model a single tendon!

btaba, 2025-04-07:
Closing since MJX implements tendons now.

closed 2025-04-07 by btaba