google/brax #150: Installation issues

HowdyMoto, 2022-01-15:
Following installation instructions, on a pretty fresh install of OSX with an M1 Pro, I get:
ERROR: Cannot install brax==0.0.10, brax==0.0.2, brax==0.0.3, brax==0.0.4, brax==0.0.5, brax==0.0.6, brax==0.0.7, brax==0.0.8 and brax==0.0.9 because these package versions have conflicting dependencies.

The conflict is caused by:
    brax 0.0.10 depends on jaxlib
    brax 0.0.9 depends on jaxlib
    brax 0.0.8 depends on jaxlib
    brax 0.0.7 depends on jaxlib
    brax 0.0.6 depends on jaxlib
    brax 0.0.5 depends on jaxlib
    brax 0.0.4 depends on jaxlib
    brax 0.0.3 depends on jaxlib
    brax 0.0.2 depends on jaxlib

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies

I installed jax with:
pip install --upgrade pip
pip install --upgrade "jax[cpu]"

...and re-ran the brax install, and still got the above error.

erikfrey, 2022-01-20:
Hi there!  Jaxlib support on the M1 is quite new, so it's possible you're running into a problem there.

There should be a pre-built wheel for M1 now, but it may be that you aren't finding it.  Can you try `pip install jaxlib` and see if that works? (surprisingly, `pip install jax` does not install jaxlib)  If it does not work, see this thread for more info:

https://github.com/google/jax/issues/5501

HowdyMoto, 2023-04-22:
Works fine now - you can close this out.

closed 2023-04-22 by HowdyMoto