google/brax #151: Ball on fixed ground starts bouncing

ymd-h, 2022-01-15:
Hi, I encountered strange situation.

A ball on the fixed ground starts bouncing.
Moreover, energy is not conserved. (It becomes larger and larger.)

```python
import matplotlib.pyplot as plt
import numpy as np
import jax

try:
  import brax
except ImportError:
  from IPython.display import clear_output 
  !pip install git+https://github.com/google/brax.git@main
  clear_output()
  import brax

RADIUS = 0.1
SIM_STEPS = 300

bouncy_ball = brax.Config(dt=0.01, substeps=4)

# ground is a frozen (immovable) infinite plane
ground = bouncy_ball.bodies.add(name='ground')
ground.frozen.all = True
plane = ground.colliders.add().plane
plane.SetInParent()

ball = bouncy_ball.bodies.add(name='ball', mass=1.0)
cap = ball.colliders.add().capsule
cap.radius, cap.length = RADIUS, 2*RADIUS

bouncy_ball.gravity.z = -9.8
bouncy_ball.elasticity = 1.0

sys = brax.System(bouncy_ball)
step_compiled = jax.jit(sys.step)

def reset_qp():
  return brax.QP(
    pos = np.array([[0., 0., 0.],    # ground
                    [0., 0., RADIUS]]), # ball is on ground
    vel = np.array([[0., 0., 0.],       # ground
                    [0., 0., 0.]]),     # ball
    rot = np.array([[1., 0., 0., 0.],   # ground
                    [1., 0., 0., 0.]]), # ball
    ang = np.array([[0., 0., 0.],       # ground
                    [0., 0., 0.]])      # ball
)

qp = reset_qp()
qps = [qp]
for _ in range(SIM_STEPS):
    qp, _ = step_compiled(qp, [])
    qps.append(qp)

qz = [_qp.pos[1, 2] for _qp in qps]
vz = [_qp.vel[1, 2] for _qp in qps]

plt.figure(figsize=(5,5))
ax = plt.gca()
ax.plot(qz, marker=".", linestyle=":", label="Z (left)")

ax2 = ax.twinx()
ax2.plot(vz, marker=".", linestyle=":", label="Vz (right)", color="tab:orange")

ax.legend(*[a+b for a,b in zip(ax.get_legend_handles_labels(), ax2.get_legend_handles_labels())])
plt.show()
```

[image]

cdfreeman-google, 2022-02-08:
Yep, this is a conspiracy of a few different approximations biting the simulator simultaneously.

tldr: You can avoid this by using elasticity slightly less than 1, and by cranking up the number of substeps.  At elasticity _exactly_ 1.0, the error accumulated by the ball slightly penetrating the ground after every substep allows it to smuggle energy out of the simulator.  When elasticity is less than 1, it can still smuggle a little energy, but it's at least bounded.  Here's a version with elasticity .99 and substeps = 16:

[image]

This is a known issue with discrete physics simulation.  Fancier simulators implement "sleeping" methods, where objects that are at rest are simply not simulated to avoid the accumulation of this type of error.

Thankfully, this sort of thing should disappear completely in a new version of Brax that's in the works right now :)

ymd-h, 2022-02-08:
@cdfreeman-google 
Thank you for your reply.
I understand the situation and try to use smaller elasticity.

> Thankfully, this sort of thing should disappear completely in a new version of Brax that's in the works right now :)

Great!
I hope it will come soon.

closed 2022-02-08 by ymd-h