google/brax #157: Inverse mapping of Joint angle_vel

alexoshin, 2022-01-29:
Hi all,

I was wondering if there was a method for doing the inverse of the Joint angle_vel method, i.e., go from DOF position and velocities back to a QP. These should be a 1-1 mapping as long as you have access to the root states. It seems like its performed when getting the default QP of a system here:
https://github.com/google/brax/blob/a09f3e75390c1ff017565ba2acfa77e838535fd3/brax/physics/system.py#L119-L180

It might be nice to have that as a dedicated method, since this gives a more succinct representation of the state. Let me know if I missed something somewhere.

Thanks!

erikfrey, 2022-02-06:
Yes!  We have a function called `angle_vel` that can see used in humanoid here:

https://github.com/google/brax/blob/a09f3e75390c1ff017565ba2acfa77e838535fd3/brax/envs/humanoid.py#L81

100% agree that it would make sense to organize the code such that there are functions that live side by side that convert from maximal to generalized coordinates and back again.  Forward kinematics / inverse kinematics, and it would probably clean up the code in `default_qp` a fair bit.  I'd be happy to review a PR if someone would like to give this a go.

namheegordonkim, 2023-04-20:
Hey, just checking in to see if anyone figured this one out; with RL methods requiring simulator reset to a particular state, being able to reconstruct states from the aforementioned succinct representations is quite important.

@erikfrey In light of the major rework from a few weeks back, does the new version of Brax support this conversion, i.e., going from qpos and qvel to full state of body links?

namheegordonkim, 2023-04-20:
Will also add that the workaround is to represent robot state by concatenating all of this body links' pos, rot, vel, and ang into one big vector to represent the state, but obviously this isn't optimal.