google/brax #163: [Question] [Enhancement] Aerodynamics

AlessandroZavoli, 2022-02-15:
Is it possible to add custom aerodynamics to an object, e.g., to model wing-based objects like airplanes or helicopters?

soldierofhell, 2022-02-16:
Do you mean something like [lift&drug plugin](https://gazebosim.org/tutorials?tut=aerodynamics&cat=physics) in gazebo?

AlessandroZavoli, 2022-02-16:
Yes, something like that. I could provide the mathematical equations for a few cases of interest (e.g., flat plate with no lift, 2D airfoil) or a generic model in terms of classical aerodynamic coefficients.

This is probably lower in priority than [issues#156](https://github.com/google/brax/issues/156), that is, fixing second-order terms in the Euler equations for attitude dynamics.