google/brax #167: auto_reset=True produces environment without randomness

jjyyxx, 2022-02-22:
Currently, when creating an environment, `auto_reset` defaults to True. However, when digging into the `AutoResetWrapper` class, I found that the `auto_reset` behavior simply resets the environment to `first_qp` and `first_obs`, which is the initial state after the last manual reset. Thus, `auto_reset` repeats the same initial state for every episode, which may have a negative impact on training. I wonder if such behavior is intended, or did I misunderstand something?

cdfreeman-google, 2022-02-22:
Interesting! Could you provide a repro of this behavior?  If you're referring to these lines:

```
def reset(self, rng: jp.ndarray) -> brax_env.State:
    state = self.env.reset(rng)
    state.info['first_qp'] = state.qp
    state.info['first_obs'] = state.obs
    return state
```
All that is happening here is that `state` is recording the first qp of the trajectory and the first observation of the trajectoryi n the `info` dict so that it can be used elsewhere.  `state` is properly being reset here, as far as I can tell

jjyyxx, 2022-02-23:
Thanks for your reply. But actually, my question concerns the `step` method. In my understanding, `AutoReset` means that when a Brax env is done, it is automatically reset, so that a user does not need to call `reset` manually (except for the initial reset). For a vectorized env, `AutoReset` is a very convenient wrapper. However, instead of properly resetting env with randomness, the `step` method simply reset the environment to `first_qp` and `first_obs` you mentioned in the `reset` method, and the wrapped env's `reset` method is never called.

A minimal repro would be
```python
>>> import brax.envs
>>> env = brax.envs.create_gym_env('inverted_pendulum', episode_length=3)
>>> env.seed(0)
>>> env.reset()
DeviceArray([ 1.000000e-01,  1.000000e+00,  0.000000e+00,  0.000000e+00,
              0.000000e+00,  9.823757e-03,  0.000000e+00,  0.000000e+00,
              0.000000e+00, -6.298446e-04], dtype=float32)
>>> env.reset()  # Different from above, as expected
DeviceArray([ 0.1       ,  1.        ,  0.        ,  0.        ,
              0.        , -0.0015702 ,  0.        ,  0.        ,
              0.        ,  0.00274388], dtype=float32)
>>> env.step(env.action_space.sample() * 0.0)[0]
DeviceArray([ 1.0000000e-01,  1.0000000e+00,  0.0000000e+00,
              0.0000000e+00,  0.0000000e+00, -1.6629273e-03,
              3.5764193e-04,  0.0000000e+00, -0.0000000e+00,
              2.1486760e-03], dtype=float32)
>>> env.step(env.action_space.sample() * 0.0)[0]
DeviceArray([ 1.0000000e-01,  1.0000000e+00,  0.0000000e+00,
              0.0000000e+00,  0.0000000e+00, -1.7784013e-03,
              3.0605256e-04,  0.0000000e+00, -0.0000000e+00,
              3.9656470e-03], dtype=float32)
>>> env.step(env.action_space.sample() * 0.0)[0]  # episode_length reaches, auto reset, observation identical to last reset
DeviceArray([ 0.1       ,  1.        ,  0.        ,  0.        ,
              0.        , -0.0015702 ,  0.        ,  0.        ,
              0.        ,  0.00274388], dtype=float32)
```

erikfrey, 2022-02-25:
Hi @jjyyxx !  Yes indeed, this is by design.  Because our environments are vectorized, each step sets some variable number of environments to done.  Calculating new reset states at each step would be very expensive, so we compromise by reusing reset states.

For RL use cases with large batch sizes that we use, this does not impact training performance at all.  2048 randomly initialized environments produces enough initial randomness that it's OK if we sometimes revisit those states.

Let us know if you find evidence otherwise.  We could always introduce a mechanism to "refresh" reset states every now and then during training - like in between evals - but I doubt we need such a thing.

jjyyxx, 2022-02-26:
Thanks for your clarification. The point about batch randomness makes sense. When experimenting with halfcheetah though, `reset` takes much less time than `step`, which may not be that "expensive" (but it's on CPU device). Nevertheless, I will close the issue for now.

closed 2022-02-26 by jjyyxx