google/brax #168: Coupling bodies without joints?

namheegordonkim, 2022-02-23:
Is it possible to have two links joined together at a fixed pivot? Current workaround is to use a joint specifying parent and child and set angle limits to 0, but this incidentally increases the number of joints in the character.

erikfrey, 2022-02-25:
You can couple colliders together into a single body.  See humanoid for an example of this:

https://github.com/google/brax/blob/main/brax/envs/humanoid.py#L144

Notice the torso is a collection of colliders.  This satisfies most use cases where users want complex bodies, but please let me know if you have another use case in mind that multiple colliders per body does not satisfy.

namheegordonkim, 2022-02-25:
In fact, I made a separate issue #169 to complain about the use case where thrusters are attached to elsewhere than COM. One workaround would be to create a separate auxiliary body (not collider) on the target body and attach a thruster to the auxiliary body's COM.

erikfrey, 2022-02-25:
OK got it - so if we could support multiple thrusters at different offsets, then you would not need this?

namheegordonkim, 2022-02-25:
>if we could support multiple thrusters at different offsets, then you would not need this fixed pivot feature?

True; would not need this feature, although it would still be nice (it's consistent with e.g. MuJoCo's convention)

stephane-caron, 2022-04-06:
>  This satisfies most use cases where users want complex bodies, but please let me know if you have another use case in mind that multiple colliders per body does not satisfy.

Inertia tensors are another use case. Currently inertias are described by a mass and a diagonal matrix for the combined body, but a common pattern is that we know inertias for the colliders that the body is made of, e.g. a battery (box) inside a chassis (box).

Representing the inertia tensor by a `Vector3` is actually problematic for bodies with more than one collider. A collider's simple shape (box, capsule, sphere) does have a diagonal inertia tensor, but once several of them are combined, the centroidal inertia tensor of the resulting body may not be diagonal.

One way forward on the API side would be to move `inertia` and `mass` to `Collider`. (Then they would not be just colliders any more though.)