google/brax #169: Thruster at an offset

namheegordonkim, 2022-02-23:
Currently, the external assistive forces are applied at the center of mass, but for many shapes, it makes more sense to attach thrusters at the vertices/boundaries of the bodies instead. Supporting offsets would also allow multiple thrusters to be attached to the same body. I couldn't find anywhere if this is supported yet. If not, I'd be happy to make one and work towards a PR.

erikfrey, 2022-02-25:
This seems like a very reasonable request.  After all, joints have offsets, too.  OK, if you open a PR we'll get this in.

btaba, 2024-10-04:
Thrusters are no longer supported, please use [MJX](https://mujoco.readthedocs.io/en/stable/mjx.html) and `xfrc_applied`  :)

closed 2024-10-04 by btaba