google/brax #170: Visualization on old versions of Brax no longer supported

verityw, 2022-02-24:
I had an old version of Brax running on my computer. However, recently, the html visualization generation stopped working. I pulled the latest version and, while they work again, it seemingly broke some code I was working on (was running APG with the parameters in https://github.com/google/brax/issues/144, which used to work, but now no longer does -- previously was able to get those parameters working on ant, now ant just walks forward a few steps and stops).

However, I have an old brax install on a different computer, whose visualizations are also broken. Is there a way to update the visualizations only to work again while keeping everything else the same? What even got changed with the visualizations that ended support for old versions of Brax?

erikfrey, 2022-02-25:
Hi @verityw - how old of a version are we talking?  In very old versions of Brax, I think the visualizer html output was referencing an unversioned javascript URL, so yes that wouldn't be backwards compatible.

Newer versions of brax will pin to their sibling javascript libraries so you won't see these kinds of breakages anymore.

We might be able to help you hack your old brax install to work.  Got a colab link?

verityw, 2022-02-28:
@erikfrey I believe I cloned it on December 7th, so whatever the lastest commit back then was. Do you know what could have changed to have made the old training parameters not work as well?

I can make a colab, what stuff would you like on it?

erikfrey, 2022-03-23:
There was one big change that is unfortunately out of our control:  three.js [stopped hosting dat.gui](https://twitter.com/threejs/status/1455160906980200457?lang=en), which old scripts were referencing at [this url](https://threejs.org/examples/jsm/libs/dat.gui.module.js).

If that's the broken visualization, you could change the reference to lil-gui, as you can see we did in newer versions of `viewer.js`.

Closing this issue for now, but please reach out if this bug is still a problem.

closed 2022-03-23 by erikfrey