google/brax #171: NaNs in gradient of dynamics for humanoid

C-J-Cundy, 2022-02-24:
Hi, when experimenting with some differentiable RL algorithms I have run into a situation where I get NaNs coming out of the gradient of the dynamics function of the humanoid environment.  

I have a simple function computing the next state's observations as a function of the current action:

```
def obs_fn(a: jnp.ndarray, state):
  new_state = env.step(state, a)
  return jnp.mean(new_state.obs)
```

When I grab a big batch of states from random rollouts and evaluate grad(obs_fn) with respect to the action, on a small number of states (2 out of a batch of 4096) the resulting output is NaN.  The output of `obs_fn` itself is not NaN, so the issue seems to be coming from the gradient itself. 

The full code to replicate this is below, and also can be viewed in a colab [here](https://colab.research.google.com/drive/1QhAAtkFDKLiUip39rIJzDn_tEdl2mVZ2?usp=sharing). 

Are the NaNs here due to a bug in brax or are they the 'correct' derivative for those states? If they are 'correct' is there any smoothing or approximation supported by brax to get rid of them?  Notably these NaNs didn't appear when I only used initial states, so I imagine there could be something to do with the collisions causing them? 

I also attempted to sanity check using `jax.test_util.check_grads` but got a leaked tracer error (although I appreciate it's possible for the correct derivative through autodiff rules to be NaN and the finite-differences derivative estimate to be sensible)

PS Thanks for developing a really cool library!

<details>
  <summary>Full code listing</summary>

```
import functools
import time

from IPython.display import HTML, Image 
import gym

try:
  import brax
except ImportError:
  from IPython.display import clear_output 
  !pip install git+https://github.com/google/brax.git@main
  clear_output()
  import brax

from brax import envs
from brax import jumpy as jp
from brax.envs import to_torch
from brax.io import html
from brax.io import image
import jax
from jax import numpy as jnp
import torch
v = torch.ones(1, device='cuda')  # init torch cuda before jax

environment = "humanoid"
env = envs.create(env_name=environment)

def get_random_rollout(rng_key, T=100):
  def inner(state, rng):
    action = jax.random.uniform(rng, minval=-0.999, maxval=0.999, shape=(env.action_size,))
    next_state = env.step(state, action)
    return next_state, next_state

  _, states = jax.lax.scan(inner, env.reset(rng_key), jax.random.split(rng_key, T))
  return states

def subsample_rollout(state, rng_key, k=10):
  T = len(state.qp.pos)
  idxs = jax.random.choice(rng_key, T, (k,), replace=True)
  return jax.tree_util.tree_map(lambda x: x[idxs], state)

def collate_tree(x):
  def reshape_fn(x):
    shapes = x.shape
    return x.reshape([shapes[0] * shapes[1]] + list(shapes[2:]))
  return jax.tree_util.tree_map(reshape_fn, x)

def get_replay_buffer(n_trajectories, n_subsample, horizon):
  keys = jax.random.split(jax.random.PRNGKey(0), n_trajectories)
  keys_2 = jax.random.split(jax.random.PRNGKey(1), n_trajectories)
  rollouts = jax.jit(jax.vmap(get_random_rollout, in_axes=(0, None)), static_argnames='T')(keys, horizon)
  subsampled_rollouts = jax.jit(jax.vmap(subsample_rollout, in_axes=(0, 0, None)), static_argnames='k')(rollouts, keys_2, n_subsample)
  return collate_tree(subsampled_rollouts)

trajectories = get_replay_buffer(128, 32, 1000)

print(f"Number of state examples: {trajectories.info['steps'].shape}")

def obs_fn(a: jnp.ndarray, state):
  new_state = env.step(state, a)
  return jnp.mean(new_state.obs)

vmap_obs_fn = jax.jit(jax.vmap(obs_fn, in_axes=(0, 0)))
vmap_grad_obs_fn = jax.jit(jax.vmap(jax.grad(obs_fn, argnums=(0,)), in_axes=(0, 0)))

n_envs = len(trajectories.info['steps'])
actions = jax.random.uniform(jax.random.PRNGKey(0), minval=-0.999, maxval=0.999, shape=(n_envs, env.action_size))

obs = vmap_obs_fn(actions, trajectories)[0]

print(jnp.sum(obs))
# Gives 2.0070157

grads = vmap_grad_obs_fn(actions, trajectories)[0]

print(jnp.sum(grads))
# Gives nan
```

</details>

cdfreeman-google, 2022-03-14:
This is _probably_ due to a couple of edge case NaNs that we weren't treating carefully when taking gradients of inverse trigonometric functions.  These should be fixed in the release of Brax landing this week.  I'll ping this thread when the new changes are live.

verityw, 2022-03-25:
> This is _probably_ due to a couple of edge case NaNs that we weren't treating carefully when taking gradients of inverse trigonometric functions. These should be fixed in the release of Brax landing this week. I'll ping this thread when the new changes are live.

@cdfreeman-google Has the above fix been implemented? I'm also getting some weird NaNs when I try running APG with action repeat > 3 or so with Ant.

cdfreeman-google, 2022-03-28:
Interesting! Most of these bugs should be quashed at this point.  Would you mind sharing a repro?

verityw, 2022-03-29:
Disregard my previous post, there was a different bug in my code causing the NaNs.

closed 2022-04-08 by cdfreeman-google