google/brax #175: Training on a local GPU

karlengblom, 2022-03-14:
Hello! Thank you for your work on this project.
I'm trying to train the ant environment locally. But it's very slow as you can see, so I'm assuming it's on CPU (GPU is rtx 2070). Is it necessary to specify to use GPU like for the colab? Or something else? Grateful for advice.

(brax)  karl@omen:~$ learn ant.py
I0314 23:08:48.799718 140434552616768 metrics.py:41] Hyperparameters: {'log_frequency': 10, 'num_envs': 4, 'total_env_steps': 50000000}
I0314 23:08:48.804218 140434552616768 xla_bridge.py:247] Unable to initialize backend 'tpu_driver': NOT_FOUND: Unable to find driver in registry given worker: 
I0314 23:08:48.904914 140434552616768 xla_bridge.py:247] Unable to initialize backend 'tpu': INVALID_ARGUMENT: TpuPlatform is not available.
I0314 23:08:48.905025 140434552616768 ppo.py:219] Device count: 1, process count: 1 (id 0), local device count: 1, devices to be used count: 1
I0314 23:09:01.497590 140434552616768 ppo.py:427] starting iteration 0 12.69543743133545
I0314 23:09:07.033340 140434552616768 ppo.py:459] {'eval/episode_reward': DeviceArray(-84.00571, dtype=float32), 'eval/episode_reward_contact_cost': DeviceArray(1.6300672, dtype=float32), 'eval/episode_reward_ctrl_cost': DeviceArray(1052.7314, dtype=float32), 'eval/episode_reward_forward': DeviceArray(-3.2535172, dtype=float32), 'eval/episode_reward_survive': DeviceArray(973.6094, dtype=float32), 'eval/completed_episodes': DeviceArray(128., dtype=float32), 'eval/avg_episode_length': DeviceArray(973.6094, dtype=float32), 'speed/sps': 0, 'speed/eval_sps': 23246.63754177282, 'speed/training_walltime': 0, 'speed/eval_walltime': 5.506169557571411, 'speed/timestamp': 0}
I0314 23:09:07.040651 140434552616768 metrics.py:50] [0] eval/avg_episode_length=973.609375, eval/completed_episodes=128.0, eval/episode_reward=-84.00570678710938, eval/episode_reward_contact_cost=1.630067229270935, eval/episode_reward_ctrl_cost=1052.7314453125, eval/episode_reward_forward=-3.2535171508789062, eval/episode_reward_survive=973.609375, speed/eval_sps=23246.637542, speed/eval_walltime=5.506170, speed/sps=0, speed/timestamp=0, speed/training_walltime=0
I0314 23:44:17.747002 140434552616768 ppo.py:427] starting iteration 1 2128.9448471069336
I0314 23:44:19.419119 140434552616768 ppo.py:459] {'eval/episode_reward': DeviceArray(-1885.5591, dtype=float32), 'eval/episode_reward_contact_cost': DeviceArray(0.771517, dtype=float32), 'eval/episode_reward_ctrl_cost': DeviceArray(2502.2344, dtype=float32), 'eval/episode_reward_forward': DeviceArray(-17.747227, dtype=float32), 'eval/episode_reward_survive': DeviceArray(635.1933, dtype=float32), 'losses/entropy_loss': DeviceArray(0.30761924, dtype=float32), 'losses/policy_loss': DeviceArray(7.857656, dtype=float32), 'losses/total_loss': DeviceArray(5405., dtype=float32), 'losses/v_loss': DeviceArray(5396.834, dtype=float32), 'eval/completed_episodes': DeviceArray(150., dtype=float32), 'eval/avg_episode_length': DeviceArray(635.04663, dtype=float32), 'speed/sps': DeviceArray(2368.8562, dtype=float32), 'speed/eval_sps': 76881.63453282246, 'speed/training_walltime': 2110.698221921921, 'speed/eval_walltime': 7.171062469482422, 'speed/timestamp': 2110.698221921921}
I0314 23:44:19.421806 140434552616768 metrics.py:50] [4999920] eval/avg_episode_length=635.046630859375, eval/completed_episodes=150.0, eval/episode_reward=-1885.55908203125, eval/episode_reward_contact_cost=0.7715169787406921, eval/episode_reward_ctrl_cost=2502.234375, eval/episode_reward_forward=-17.74722671508789, eval/episode_reward_survive=635.1932983398438, losses/entropy_loss=0.30761924386024475, losses/policy_loss=7.857656002044678, losses/total_loss=5405.0, losses/v_loss=5396.833984375, speed/eval_sps=76881.634533, speed/eval_walltime=7.171062, speed/sps=2368.856201171875, speed/timestamp=2110.698222, speed/training_walltime=2110.698222

KarolisRam, 2022-03-16:
Try increasing the `num_envs` from 4 to 2048 or so (I use 2048 on a 2080 Ti). I was as confused as you when I first tried it. Not sure why the default is only 4 envs.

karlengblom, 2022-03-23:
It worked. Now it's as fast as advertised. Thank you for your help.

closed 2022-03-23 by karlengblom