google/brax #180: [Feature request] Colors (and maybe transparency) for objects

DavidSlayback, 2022-03-23:
This is likely not a high priority, but the rendering options that currently exist are hard-coded to have 3 distinct colors (using "ground" and "target" as string matches):
1) Ground is a checkerboard
2) Target is red
3) Everything else is grey

If possible, I'd like to specify an RGB value for each collider in a body and have that respected by the HTML/PyTinyRenderer renderers. Even better would be an RGBA, but I know pytinyrenderer lacks support for transparency.

An example use case would be the "ant_gather" environment from [Benchmarking Deep RL for Continuous Control](https://arxiv.org/pdf/1604.06778.pdf), which [I've reimplemented in Brax](https://colab.research.google.com/gist/DavidSlayback/562f800d145a1fa8fbe4c82c27bb9c3f/ant_gather.ipynb). 

There are 16 objects (8 apples and 8 bombs), and currently I differentiate them visually with height (bombs sunk into the ground like landmines). I'd prefer to make apples green and bombs red. Even cooler would be adding some visualization of the ant's sensor arc. I'd be interested in helping develop these options, but am not sure whether there's enough interest or where to begin

BenGutteridge, 2022-05-02:
Can I ask how you managed to color an object red? Changing its name to 'Target' for me didn't work.

erikfrey, 2022-05-02:
We now support custom colors (and custom visual meshes) as of 1e670bb359e300c298a819ad8ab17ac18e09a4cf.

If you're using the js visualizer off of cdn ( our most common use case ) then you'll have access to this in our next version bump, which should happen in the next few days.

DavidSlayback, 2022-05-04:
Thank you! Closing

closed 2022-05-04 by DavidSlayback