google/brax #182: biggym jump__cheetah doesn't work

KarolisRam, 2022-03-25:
Easiest way to demonstrate it is to go to https://colab.research.google.com/github/google/brax/blob/main/notebooks/biggym/biggym_rl.ipynb and do these two changes: `env_name = 'jump__cheetah'` and `env_params =  {}`
This will throw an error:
```
---------------------------------------------------------------------------

TypeError                                 Traceback (most recent call last)

[<ipython-input-5-62a6890dacfc>](https://localhost:8080/#) in <module>()
     11 env_fn = composer.create_fn(env_name=env_name, **env_params)
     12 # env_fn = biggym.create_fn(env_name=env_name, **env_params)
---> 13 env = env_fn()
     14 show_env(env)

[/usr/local/lib/python3.7/dist-packages/brax/experimental/composer/composer.py](https://localhost:8080/#) in create(env_name, env_desc, desc_edits, episode_length, action_repeat, auto_reset, batch_size, **kwargs)
    477     else:
    478       assert not kwargs, f'unused kwargs: {kwargs}'
--> 479     env_desc = dict(**env_desc, **desc)
    480     env_desc = composer_utils.edit_desc(env_desc, desc_edits)
    481     composer = Composer(**env_desc)

TypeError: type object argument after ** must be a mapping, not JumpCheetah
```

All other biggym envs have mappings as `ENV_DESCS` and work fine, while jump__cheetah points to a class:

[image]

cdfreeman-google, 2022-03-30:
@frt03 , could you take a look at this?

fwiw, environments in the experimental directory have no stability guarantees :)

shaneshixiang, 2022-03-30:
I am on paternity leave so there will be delays. I can take a look sometime. Or @frt03

KarolisRam, 2022-03-31:
> fwiw, environments in the experimental directory have no stability guarantees :)
I'm trying to add a similar env to biggym and I wanted to use `jump__cheetah` as a template, but now I'm not sure if it's done correctly.

frt03, 2022-04-04:
Hi @KarolisRam,

Temporally, you may change [this line](https://github.com/frt03/brax/blob/0ebb33219033d324bc56ebf80a6d6702c694951b/brax/experimental/composer/envs/__init__.py#L37) as 
```python
def is_env_desc(env_desc: Any):
  """Check if it is appropriate env_desc object."""
  return isinstance(env_desc, dict) or callable(env_desc) and (not inspect.isclass(env_desc))
```
and use `biggym.create_fn` instead of `composer.create_fn`. We may fix this in the future release.