google/brax #183: Collisions - nonphysical behaviours

esculapa, 2022-03-30:
Hi, I was trying to use Brax for simulating grasping and colliding objects. However, I could not realize my vision, since I approached several problems with Brax (starting with boxes, which did not collide, ending with pendling ball due to friction). 

I gathered my examples into a collab: https://colab.research.google.com/drive/1jmEoetm0uKFT_4JiWysUWuH6lPAWYdAL?usp=sharing .
So, my question is: am I doing something wrong, or such things are just happening in Brax and would to be eliminated in the future. I was also thinking - maybe those inaccuracies are acknowledged and accepted as the cost needed to be paid in order to allow Brax to word as intended.

cdfreeman-google, 2022-03-30:
Hello! I took a look at your colab, and you're correct that we don't yet support box-box collisions (though this is on the horizon!).  The remaining issues seem to be an artifact of your integration step size being too large.  If you increase substeps from 4 to, say, 40, you'll find the weird behaviors will go away.

esculapa, 2022-03-30:
Ok :) So I am going to check it ! I am really looking forward to box-box collisions - when (roughly) can we expect that?

esculapa, 2022-03-30:
Ok, I applied your change - however, the ball is still running faster with time...
https://colab.research.google.com/drive/1kCSvWzesmJ2bljyoL0EO3QxFcxsWYqSx?usp=sharing

cdfreeman-google, 2022-03-30:
This is actually expected behavior.  Here's the ball's x-velocity as a function of time:

[image]

When a ball hits a surface with friction, it will dissipate any rotational energy that keeps the ball's rolling out of sync with the ground---that is, until the angular velocity, \omega * Radius, is the same as its linear velocity.  Once those are the same, there is no "slipping", and thus no friction force, thus the ball will keep rolling forever.

Real balls don't do this because there are other dissipative mechanisms that make a ball stop rolling (like stickiness of the ground, and other little micro-collisions with particulates).  If you absolutely need balls to stop rolling eventually, you could alternatively add a small velocity damping term.

box-box collisions will hopefully land in the next month or two :)

cdfreeman-google, 2022-04-08:
I'm closing this thread for now.  Feel free to reopen if you continue to see issues!

closed 2022-04-08 by cdfreeman-google