google/brax #185: 'Brax Training with PyTorch on GPU' notebook fails to execute

BenGutteridge, 2022-04-07:
Due to a "can't convert cuda:0 device type tensor to numpy. Use Tensor.cpu() to copy the tensor to host memory first." error in pyplot, but fixing this raises another error.

erikfrey, 2022-05-06:
Thanks for pointing this out.  Both errors are addressed in a660b273f21f305b2d86e440f532a300775835f5 and the colab  runs again.

btaba, 2023-03-20:
Thanks for pointing this out @erwincoumans! I've fixed the brax.v2.positional import

btaba, 2023-05-02:
Hi @erwincoumans @BenGutteridge , erikfrey pushed out a new pytorch colab recently https://colab.sandbox.google.com/github/google/brax/blob/main/notebooks/training_torch.ipynb
Just tested that it runs on GPU, sorry this took a while!

closed 2023-05-02 by btaba