google/brax #187: 'jaxlib.xla_extension' has no attribute 'CpuDevice'

jzhzhang, 2022-04-18:
Thanks for your great work. I just finish the installation and the verison of the libs are:
```
brax              0.0.12 
jax                0.3.7
jaxlib             0.3.7+cuda11.cudnn805
```
However, when I run the "learn" from the README, the log shows:
```
Traceback (most recent call last):
  File "/home/yangwang/brax-0.0.12/env/bin/learn", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/home/yangwang/brax-0.0.12/bin/learn", line 4, in <module>
    from brax.training import learner
  File "/home/yangwang/brax-0.0.12/brax/training/learner.py", line 26, in <module>
    from brax.training import apg
  File "/home/yangwang/brax-0.0.12/brax/training/apg.py", line 32, in <module>
    import optax
  File "/home/yangwang/brax-0.0.12/env/lib/python3.9/site-packages/optax/__init__.py", line 17, in <module>
    from optax import experimental
  File "/home/yangwang/brax-0.0.12/env/lib/python3.9/site-packages/optax/experimental/__init__.py", line 20, in <module>
    from optax._src.experimental.complex_valued import split_real_and_imaginary
  File "/home/yangwang/brax-0.0.12/env/lib/python3.9/site-packages/optax/_src/experimental/complex_valued.py", line 32, in <module>
    import chex
  File "/home/yangwang/brax-0.0.12/env/lib/python3.9/site-packages/chex/__init__.py", line 17, in <module>
    from chex._src.asserts import assert_axis_dimension
  File "/home/yangwang/brax-0.0.12/env/lib/python3.9/site-packages/chex/_src/asserts.py", line 26, in <module>
    from chex._src import asserts_internal as _ai
  File "/home/yangwang/brax-0.0.12/env/lib/python3.9/site-packages/chex/_src/asserts_internal.py", line 32, in <module>
    from chex._src import pytypes
  File "/home/yangwang/brax-0.0.12/env/lib/python3.9/site-packages/chex/_src/pytypes.py", line 40, in <module>
    CpuDevice = jax.lib.xla_extension.CpuDevice
AttributeError: module 'jaxlib.xla_extension' has no attribute 'CpuDevice'
```
I do not have a clue about this problem and I already use the latest version of both brax and jax (with GPU option). 

Hope you can give me some suggestion.

jzhzhang, 2022-04-18:
Here is output of "python -c 'import jax; print(jax.devices())'"
```
[GpuDevice(id=0, process_index=0), GpuDevice(id=1, process_index=0), GpuDevice(id=2, process_index=0), GpuDevice(id=3, process_index=0), GpuDevice(id=4, process_index=0), GpuDevice(id=5, process_index=0), GpuDevice(id=6, process_index=0), GpuDevice(id=7, process_index=0)]
```

proceduralia, 2022-04-18:
Hi,

I think this is due to an outdated version of `chex`. To temporarily solve the problem I did:

```
pip uninstall chex
pip install git+https://github.com/deepmind/chex.git --upgrade --no-cache
```

Hope it helps,

Pierluca

jzhzhang, 2022-04-18:
@proceduralia  Hi, thanks for your quick response. 

We follow your suggestion, and reinstall the chex with version v0.1.2. Unfortunately, we still has the ```no attribute 'Cpu Device' problem```.

BTW, we can use some brax function without any problem,e.g. ```brax.Config(dt=0.05, substeps=20, dynamics_mode='pbd' )``` . This may only happens in the ```learn program```.

Yi-FanLi, 2022-04-18:
Thank you, Pierluca.
I also incurred the same issue, and it works.

erikfrey, 2022-04-19:
Thanks for pointing this out!  It looks like chex v0.1.3 was just released - it addresses this bug:

https://github.com/deepmind/chex/releases/tag/v0.1.3

 Please let us know if you're still having any issues after upgrading to the latest chex.

samyakai, 2022-04-21:
@erikfrey I am also facing the same issue. So I upgraded as you suggested but now I am facing this error: "AttributeError: module 'jax.random' has no attribute 'KeyArray'"

erikfrey, 2022-04-26:
@samyakai what is your Jax version?  You may need to upgrade it - I think KeyArray was introduced some months back.  I can confirm it's there in latest Jax.

jzhzhang, 2022-04-29:
@erikfrey @proceduralia  The latest version works great! Really appreciate your help.

closed 2022-04-29 by jzhzhang