google/brax #191: Joint of Reacher doesn't move

Markus28, 2022-04-22:
If we use Reacher in the provided notebook https://colab.research.google.com/github/google/brax/blob/main/notebooks/environments.ipynb the first joint doesn't move. Maybe I'm doing something wrong, but this looks like a bug to me.

erikfrey, 2022-04-25:
@Markus28 yes indeed this is a bug!  We're looking into it now - thanks for the heads up.

erikfrey, 2022-05-06:
Ah this wasn't a bug so much as the actuation signal I had chosen was canceling out the torques on the first joint.  I've updated the colab - if you rerun it with reacher you'll see both joints rotating now.

btaba, 2022-06-14:
Thanks @Markus28  for reporting, closing the issue for now since reacher seems to work in the colab

closed 2022-06-14 by btaba