google/brax #198: [Question] How to debug training algorithms?

vuoristo, 2022-05-26:
Hi,

I'm trying to debug the PPO algorithm from `brax.training.ppo` while disabling jax jitting. This causes an error with jumpy trying to call a numpy method a Traced jax array. See the command line and log below. Is there a way to run this without jit turned on?

```bash
$ JAX_DISABLE_JIT=1 python -m brax.training.learner
I0526 20:02:54.884144 140223632732992 metrics.py:42] Hyperparameters: {'log_frequency': 10, 'num_envs': 4, 'total_env_steps': 50000000}
I0526 20:02:54.966016 140223632732992 xla_bridge.py:330] Unable to initialize backend 'tpu_driver': NOT_FOUND: Unable to find driver in registry given worker:
I0526 20:02:57.443377 140223632732992 xla_bridge.py:330] Unable to initialize backend 'tpu': INVALID_ARGUMENT: TpuPlatform is not available.
I0526 20:02:57.444162 140223632732992 ppo.py:221] Device count: 8, process count: 1 (id 0), local device count: 8, devices to be used count: 8
Traceback (most recent call last):
  File "/data/tucana/risrio/brax/brax/training/learner.py", line 229, in <module>
    app.run(main)
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/absl/app.py", line 312, in run
    _run_main(main, args)
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/absl/app.py", line 258, in _run_main
    sys.exit(main(argv))
  File "/data/tucana/risrio/brax/brax/training/learner.py", line 163, in main
    inference_fn, params, _ = ppo.train(
  File "/data/tucana/risrio/brax/brax/training/ppo.py", line 243, in train
    first_state = reset_fn(key_envs)
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/jax/_src/traceback_util.py", line 162, in reraise_with_filtered_traceback
    return fun(*args, **kwargs)
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/jax/_src/api.py", line 1485, in vmap_f
    out_flat = batching.batch(
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/jax/linear_util.py", line 168, in call_wrapped
    ans = self.f(*args, **dict(self.params, **kwargs))
  File "/data/tucana/risrio/brax/brax/envs/wrappers.py", line 87, in reset
    state = self.env.reset(rng)
  File "/data/tucana/risrio/brax/brax/envs/wrappers.py", line 66, in reset
    state = self.env.reset(rng)
  File "/data/tucana/risrio/brax/brax/envs/ant.py", line 35, in reset
    qp = self.sys.default_qp(joint_angle=qpos, joint_velocity=qvel)
  File "/data/tucana/risrio/brax/brax/physics/system.py", line 160, in default_qp
    _, (local_rot, local_ang) = jp.scan(local_rot_ang, (), xs, len(joint))
  File "/data/tucana/risrio/brax/brax/jumpy.py", line 113, in scan
    stacked_y = jax.tree_map(lambda *y: onp.vstack(y), *maybe_reversed(ys))
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/jax/_src/tree_util.py", line 184, in tree_map
    return treedef.unflatten(f(*xs) for xs in zip(*all_leaves))
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/jax/_src/tree_util.py", line 184, in <genexpr>
    return treedef.unflatten(f(*xs) for xs in zip(*all_leaves))
  File "/data/tucana/risrio/brax/brax/jumpy.py", line 113, in <lambda>
    stacked_y = jax.tree_map(lambda *y: onp.vstack(y), *maybe_reversed(ys))
  File "<__array_function__ internals>", line 180, in vstack
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/numpy/core/shape_base.py", line 279, in vstack
    arrs = atleast_2d(*tup)
  File "<__array_function__ internals>", line 180, in atleast_2d
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/numpy/core/shape_base.py", line 121, in atleast_2d
    ary = asanyarray(ary)
  File "/data/tucana/risrio/miniconda3/envs/braxbox/lib/python3.9/site-packages/jax/core.py", line 534, in __array__
    raise TracerArrayConversionError(self)
jax._src.traceback_util.UnfilteredStackTrace: jax._src.errors.TracerArrayConversionError: The numpy.ndarray conversion method __array__() was called on the JAX Tracer object Traced<ShapedArray(float32[4])>with<BatchTrace(level=1/0)> with
  val = DeviceArray([[ 9.9932128e-01, -4.0896823e-18,  0.0000000e+00,
              -3.6836583e-02],
             [ 9.9992150e-01,  1.3911164e-18,  0.0000000e+00,
               1.2530062e-02],
             [ 9.9997675e-01,  7.5738963e-19,  0.0000000e+00,
               6.8219593e-03],
             [ 9.9998224e-01,  6.6165040e-19,  0.0000000e+00,
               5.9596170e-03],
             [ 9.9981582e-01,  2.1306611e-18,  0.0000000e+00,
               1.9191289e-02],
             [ 9.9890524e-01,  5.1935134e-18,  0.0000000e+00,
               4.6779010e-02],
             [ 9.9945801e-01,  3.6548429e-18,  0.0000000e+00,
               3.2919899e-02],
             [ 9.9999464e-01, -3.6343356e-19,  0.0000000e+00,
              -3.2735185e-03]], dtype=float32)
  batch_dim = 0
This Tracer was created on line /data/tucana/risrio/brax/brax/jumpy.py:525 (array)
See https://jax.readthedocs.io/en/latest/errors.html#jax.errors.TracerArrayConversionError
...
```

I'm running on c7735f34a48c3499516c3359d016057ed653f810 with the following packages installed in a conda environment.

```
Package            Version              Location
------------------ -------------------- ------------------------
absl-py            1.0.0
brax               0.0.13               /data/tucana/risrio/brax
certifi            2022.5.18.1
chex               0.1.3
cloudpickle        2.1.0
cycler             0.11.0
dataclasses        0.6
dm-tree            0.1.7
flatbuffers        2.0
flax               0.5.0
fonttools          4.33.3
grpcio             1.46.3
gym                0.23.1
gym-notices        0.0.6
importlib-metadata 4.11.4
jax                0.3.13
jaxlib             0.3.8+cuda11.cudnn82
kiwisolver         1.4.2
matplotlib         3.5.2
msgpack            1.0.3
numpy              1.22.4
opt-einsum         3.3.0
optax              0.1.2
packaging          21.3
Pillow             9.1.1
pip                21.2.4
protobuf           3.20.1
pyparsing          3.0.9
python-dateutil    2.8.2
pytinyrenderer     0.0.13
scipy              1.8.1
setuptools         61.2.0
six                1.16.0
tensorboardX       2.5
toolz              0.11.2
trimesh            3.12.5
typing_extensions  4.2.0
wheel              0.37.1
zipp               3.8.0
```

vuoristo, 2022-05-27:
After some more digging, I think the problem is that `jumpy` cannot know whether it should use `jax.numpy` or `numpy` when jit is disabled. I couldn't figure out a way to make `jumpy` behave consistently with the jit turned on and off. Below are a couple of cases where `jumpy` cannot decide correctly.

On `brax/physics/system.py:160` some of the inputs to the scan can be of type `jax.interpreters.batching.BatchTracer` even when jitting is disabled, which will break the `numpy` based alternative in `jp.scan`.

On `brax/physics/system.py:176`. When jit is disabled, the arrays inside the `set_qp` function are not automatically made into jax arrays, and calling `jp.index_update` will not work. When jit is on, scanning over the inputs turns them into jax arrays.

Also, you cannot just force `jumpy` to always use `jax.numpy` for fixing these cases, because for example on `brax/physics/colliders.py:1175` you will get `TypeError: unhashable type: 'DeviceArray'` error if `col_a.body.idx` was initialized with jax.

erikfrey, 2022-05-27:
@vuoristo thanks for pointing this out.  This is a scenario where jumpy is perhaps more trouble than benefit, and you're correct that it does not always have the right context for whether to use jax.numpy or numpy.

It may make sense to have a flag or config/env that tells jumpy to use jax even when running unjitted.  If you'd like to open a PR for this we'd be happy to take a look, otherwise it is on our TODO list.

But frankly, since we added the jumpy shim, jax jit time has gotten better, particularly on CPU... so we may just remove jumpy at some point in the future.

vuoristo, 2022-05-28:
Made a pull request which adds the environment variable and makes a couple of other required changes. I'm not super familiar with the code base yet so I'm unsure whether these changes have any implications beyond the immediate ones.

btaba, 2022-06-14:
Thanks for the PR @vuoristo! Closing the issue for now since it was merged