google/brax #201: Memory error on some environments when using ensemble methods

cvoelcker, 2022-06-13:
Hi, I have an extremely strange bug that is slightly hard to reproduce :)
I am trying to integrate brax with SAC and want to use an ensemble version of the algorithm. When I use the ensemble, the brax environments (tested so far on halfcheetah, walker2d, grasp and hopper) raise the following error. The original SAC version doe not! In addition, the ant environment for some reason works as well.

I am wrapping the torch code in both cases with torch.no_grad() and detach from the tensor. The shapes going into brax are the same in either case. I think this is some strange memory allocation problem that happens when torch (internally?) duplicates the Tensor to compute the ensembled predictions, but that is pure conjecture.

Would be very happy about any hints.

```
Traceback (most recent call last):
   File "sac/main.py", line 80, in main
    next_state, reward, done, _ = env.step(action)  # Step
  File "sac/env/wrappers.py", line 104, in step
    return self.env.step(action)
  File "brax/envs/to_torch.py", line 59, in step
    obs, reward, done, info = super().step(action)
  File "lib/python3.10/site-packages/gym/core.py", line 289, in step
    return self.env.step(action)
  File "python3.10/site-packages/gym/wrappers/order_enforcing.py", line 11, in step
    observation, reward, done, info = self.env.step(action)
  File "python3.10/site-packages/brax/envs/wrappers.py", line 283, in step
    self._state, obs, reward, done, info = self._step(self._state, action)
ValueError: INTERNAL: Address of buffer 11 must be a multiple of 10, but was 0x7f588c3d0848
````

lebrice, 2022-06-14:
Hey there @cvoelcker ! (I'm not a Brax maintainer btw, I just wrote the jax2torch conversion functions).

> I think this is some strange memory allocation problem that happens when torch (internally?) duplicates the Tensor to compute the ensembled predictions, but that is pure conjecture.

The Jax->Torch conversion is done via their respective `dlpack` interfaces, which essentially makes it possible to read the same tensor from either Jax or Torch, in-place, on the GPU.

The conversions are done:
- [here](https://github.com/google/brax/blob/9d6b7ced2a13da0d074b5e9fbd3aad8311e26997/brax/io/torch.py#L49) for torch -> Jax
- [here](https://github.com/google/brax/blob/9d6b7ced2a13da0d074b5e9fbd3aad8311e26997/brax/io/torch.py#L80) for Jax -> Torch

This seems a bit tough to debug without any code though. Would you mind sharing some of the code that does the "ensembled predictions"?
I've never personally seen this kind of error before.

lebrice, 2022-06-28:
> Wrapping action in torch.clone fixes the error interestingly enough, 

Then it sounds like what I was suspecting might be correct.