google/brax #209: Inverted pendulum keeps resetting its state

mhr, 2022-07-20:
Hello! Brax is really cool, but I've run into a minor issue with the inverted pendulum example. I'm probably misunderstanding something.

Why, when I give the environment a torque of 1 repeatedly, does the cart not move continuously in one direction? Instead, it resets its state every few timesteps.

Here is an example notebook illustrating the behavior: https://colab.research.google.com/drive/1omEuqhHad2CyI23ar9oT4hECsjuRRb9P#scrollTo=ssCOanHc8JH_

Thank you!

erikfrey, 2022-07-27:
@mhr hello!  The inverted pendulum reward function is quite harsh, by design.  If the pole begins to fall even a bit (past 0.2 radians), the environment is done.  You can see that here:

https://github.com/google/brax/blob/main/brax/envs/inverted_pendulum.py#L147

Also, please note that the actions in this environment don't correspond to torque, but instead linear force applied to the cart, with magnitude representing the amount of force and sign representing the direction.

mhr, 2022-07-27:
Thanks! And thank you for correcting me WRT torque.

closed 2022-07-27 by mhr