google/brax #213: suspected to found a bug

fangtiancheng, 2022-08-02:
```python
@pytree.register
class TwoWayCollider(Collider):
  """Calculates two-way impulses on collidable pairs."""

  def _contact(self, col_a: geometry.Collidable, col_b: geometry.Collidable,
               qp_a: QP, qp_b: QP,
               contact: geometry.Contact) -> Tuple[P, Optional[P]]:
    """Calculates impulse on a body due to a contact."""
    # there are a few ways to combine material properties during contact.
    # multiplying is a reasonable default.  in the future we may allow others
    elasticity = col_a.elasticity * col_b.elasticity
    friction = col_a.friction * col_b.friction
    rel_pos_a = contact.pos - qp_a.pos
    rel_pos_b = contact.pos - qp_b.pos
    baumgarte_vel = self.baumgarte_erp * contact.penetration
    normal_vel = jp.dot(contact.normal, contact.vel)
    temp1 = col_a.body.inertia * jp.cross(rel_pos_a, contact.normal)
    temp2 = col_b.body.inertia * jp.cross(rel_pos_b, contact.normal)
    ang = jp.dot(contact.normal,
                 jp.cross(temp1, rel_pos_a) + jp.cross(temp2, rel_pos_b))
    impulse = (-1. * (1. + elasticity) * normal_vel + baumgarte_vel) / (
        (1. / col_a.body.mass) + (1. / col_b.body.mass) + ang)
    dp_n_a = col_a.body.impulse(qp_a, impulse * contact.normal, contact.pos)
    dp_n_b = col_b.body.impulse(qp_b, -impulse * contact.normal, contact.pos)

    # apply drag due to friction acting parallel to the surface contact
    vel_d = contact.vel - normal_vel * contact.normal
    impulse_d = jp.safe_norm(vel_d) / ((1. / col_a.body.mass) +
                                       (1. / col_b.body.mass) + ang)
    # drag magnitude cannot exceed max friction
    impulse_d = jp.minimum(impulse_d, friction * impulse)
    dir_d = vel_d / (1e-6 + jp.safe_norm(vel_d))
    dp_d_a = col_a.body.impulse(qp_a, -impulse_d * dir_d, contact.pos)
    dp_d_b = col_a.body.impulse(qp_b, impulse_d * dir_d, contact.pos)
    # apply collision normal if penetrating, approaching, and oriented correctly
    apply_n = jp.where(
        (contact.penetration > 0.) & (normal_vel < 0) & (impulse > 0.), 1., 0.)
    # apply drag if moving laterally above threshold
    apply_d = apply_n * jp.where(jp.safe_norm(vel_d) > 0.01, 1., 0.)

    dp_a = dp_n_a * apply_n + dp_d_a * apply_d
    dp_b = dp_n_b * apply_n + dp_d_b * apply_d
    return dp_a, dp_b
```
This is your code in `brax/physics/colliders.py:418`. Notice that in line 450:
```python
    dp_d_a = col_a.body.impulse(qp_a, -impulse_d * dir_d, contact.pos)
    dp_d_b = col_a.body.impulse(qp_b, impulse_d * dir_d, contact.pos)
    #        ^^^^^
```
This formula does not have good symmetry between a and b.

btaba, 2022-08-02:
Thanks @fangtiancheng for finding and reporting the bug! We'll have a fix out soon

btaba, 2022-08-02:
Hi @fangtiancheng we just pushed a fix, thank you again for pointing this out, and thanks for using Brax!!

closed 2022-08-02 by btaba