google/brax #217: A little question

fangtiancheng, 2022-08-29:
```python
      dp_p = P(
          vel=p / col_a.body.mass,
          ang=jp.cross(col_a.body.inertia * (contact.pos - qp_a.pos), p))

      dp_c = P(
          vel=-p / col_a.body.mass,
          ang=jp.cross(col_b.body.inertia * (contact.pos - qp_b.pos), -p))
```
This is your code in `brax/physics/colliders` from line 648 to line 654. 
Notice that in line 653:
```python
          vel=-p / col_a.body.mass,
                   ^^^^^
```
This place does not satisfy symmetry. Buf if I change `col_a` to `col_b`, the physics_test.BoxCapsuleTest will fail. Is there a bug in this place?

fangtiancheng, 2022-08-29:
Run the following compact test:
```python
# add this function to class QP
def __repr__(self):
  ans = ''
  ans += "mPos\t"
  for x in self.pos:
    ans += str(x) + '\t'
  ans += '\n'
  ans += "mVel\t"
  for x in self.vel:
    ans += str(x) + '\t'
  return ans

class BoxCapsuleTest(absltest.TestCase):
  _CONFIG = """
    dt: 0.05 substeps: 30 friction: 1
    gravity { z: -9.8 }
    bodies {
      name: "box2" mass: 10
      colliders { box { halfsize { x: 0.5 y: 0.5 z: 0.5 }}}
      inertia { x: 1 y: 1 z: 1 }
    }
    bodies {
      name: "capsule2" mass: 1
      colliders { capsule { length: 2 radius: 0.2 } }
      inertia { x: 1 y: 1 z: 1 }
    }
    bodies {
      name: "box3"
      colliders {
        box {
          halfsize: { x: 0.5 y: 0.5 z: 0.5 }
        }
      }
      mass: 1.0
      frozen { all: true }
    }
    bodies { name: "Ground" frozen: { all: true } colliders { plane {}}}
    defaults {
      qps {
        name: "capsule2"
        pos { x: 2 y: 0 z: 1 }
      }
      qps {
        name: "box2"
        pos { x: 2 y: 0 z: 3.5 }
      }
    }
    solver_scale_collide: .3
  """

  def test_box_hits_capsule(self):
    """A box falls onto a capsule and stays above it."""
    sys = brax.System(text_format.Parse(BoxCapsuleTest._CONFIG, brax.Config()))
    qp = sys.default_qp()
    step_fn = jax.jit(sys.step)
    for i in range(30):
      qp, _ = step_fn(qp, jp.array([]))
      print("\n----i = %d----"%(i, ))
      print(qp)
```

After changing col_a to col_b, you will get:
```
.......
----i = 7----
mPos	[2.        0.        2.7139626]	[2.        0.        0.9998728]	[0.  0.  0.5]	[0. 0. 0.]	
mVel	[ 0.         0.        -3.9138794]	[0.         0.         0.00815392]	[0. 0. 0.]	[0. 0. 0.]	

----i = 8----
mPos	[2.      0.      2.50563]	[2.         0.         0.99987286]	[0.  0.  0.5]	[0. 0. 0.]	
mVel	[ 0.         0.        -4.4031143]	[0.         0.         0.00815392]	[0. 0. 0.]	[0. 0. 0.]	

----i = 9----
mPos	[2.000000e+00 7.839667e-09 2.457123e+00]	[2.0000000e+00 2.8130854e-08 9.7893924e-01]	[0.  0.  0.5]	[0. 0. 0.]	
mVel	[3.6502911e-07 1.9645955e-07 2.7581680e+00]	[-4.1942858e-06  2.9314392e-06  1.3593559e+00]	[0. 0. 0.]	[0. 0. 0.]	

----i = 10----
mPos	[2.0000000e+00 1.8923261e-08 2.6096663e+00]	[2.0000000e+00 1.8683544e-07 1.0499157e+00]	[0.  0.  0.5]	[0. 0. 0.]	
mVel	[0.0000000e+00 2.0786253e-07 2.8638840e+00]	[0.0000000e+00 3.3759961e-06 1.2125015e+00]	[0. 0. 0.]	[0. 0. 0.]	
.......
```
The collision occurred between `i = 8` and `i = 9`.

erikfrey, 2022-09-02:
Wow, good catch!  Would you like to submit a PR with the fix?  It does break a test, but the test needs to be changed after this fix, as the box is now bouncing more as a result of the corrected mass ratio:

[image]

So this is very good.  I would suggest this as the change to get the test to pass again:

```diff
- # box falls on capsule with non-unit mass ratio
+ # box falls on capsule with non-unit mass ratio, bounces a bit
- self.assertEqual(qp.pos[2, 2], 2.5, 2)
+ self.assertGreaterEqual(qp.pos[2, 2], 2.5, 2)
```

If would like to open a PR with your fix, please do.  Otherwise we will fix this in the next push.

closed 2022-09-16 by fangtiancheng