google/brax #219: Brax 0.1.13 breaks with JAX 0.3.16+

cemlyn007, 2022-09-03:
Hey 👋,

First of all thank you guys for working on Brax, really cool library!

Small incompatibility issue, JAX have deprecated `tree_multimap` in favour of `tree_map`. See [here](https://github.com/google/jax/releases/tag/jax-v0.3.16) for more.

Steps to reproduce:
```
pip install jax==0.3.16 brax==0.1.13
learn
```
Produces:
```
AttributeError: module 'jax' has no attribute 'tree_multimap'
```

I am happy to submit a PR.

cemlyn007, 2022-09-03:
Sorry I see that in the main branch this has been fixed.

Do you guys know when you will make another version release to PyPi?

btaba, 2022-09-09:
Hi @cemlyn007 thanks for using brax and bringing up this issue, we just pushed a PyPi release (0.0.15). Let us know if you have any other issues!

closed 2022-09-09 by btaba

cemlyn007, 2022-09-10:
Hi @btaba, thank you very much guys! Hope you all have a good weekend. 👋