google/brax #222: Add Brax to conda

pseudo-rnd-thoughts, 2022-09-12:
As a number of machine learning projects use conda and it is better for conda is all dependencies are also on conda

Could Brax to added to conda-forge?

erikfrey, 2022-09-12:
That's a great suggestion.  Is this difficult to do?  We publish to PyPI via github actions - would be great if we could support conda in the same way.

If you provide some pointers we can look into it, or better yet if you know of someone who could get it going for us, we could probably maintain it afterwards.

pseudo-rnd-thoughts, 2022-09-12:
I haven't got much experience of using it either, but we are planning on adding it to Gym
I know that mujoco and jax are both of conda already

This is a starting point https://github.com/conda-forge/staged-recipes
Once a recipe has been made, then conda checks pypi and will update the conda release for each new release (unless there are issues)

traversaro, 2022-09-13:
Hello everyone, I mantain several robotics-related packages on conda-forge, and I will be happy to kick-start the brax recipe. I opened a PR in https://github.com/conda-forge/staged-recipes/pull/20411/files, and I listed @erikfrey as additional mantainer. Creating the PR was relatively simple as I simply used grayskull at https://www.marcelotrevisani.com/grayskull (code: https://github.com/conda-incubator/grayskull), that is a conda recipe generator that simply takes the metadata from PyPI. 

As @pseudo-rnd-thoughts mentions, there is an automatic system that checks when a new release reaches PyPI, and automatically creates a PR to update the version of brax on conda-forge. 

@erikfrey If you are ok with that, feel free to comment in  https://github.com/conda-forge/staged-recipes/pull/20411/files to say that you are ok in being a mantainer of the package.

traversaro, 2022-09-13:
I was a bit too optimistic, apparently a dependency (`pytinyrenderer`) is missing, probably we need to first package that on conda-forge.

traversaro, 2022-09-13:
`pytinyrenderer` has some tricky points (C++ dependencies are vendored instead of found in the system, legacy CMake ways of finding Python are used) that complexify it being package in conda-forge. I can do it in the future, but it is not something I can handle quickly.

erikfrey, 2022-09-13:
@traversaro and yes happy to leave a comment on that PR, but before I do, can you give me a quick sense of what kind of future maintenance I'm signing up for?

traversaro, 2022-09-14:
> @traversaro and yes happy to leave a comment on that PR, but before I do, can you give me a quick sense of what kind of future maintenance I'm signing up for?

In general, I think that pure Python packages are relatively easily to mantain, as there is no "compilation step" that can go wrong. Also, bumping the version is completly automated, see for example the PRs for the `evojax` feedstock: https://github.com/conda-forge/evojax-feedstock/pulls?q=is%3Apr+is%3Aclosed (the one that `[bot-automerge]` were both created and merged by bots).  The typical sources of problem are:
* Dependencies have not been upgraded, and so the `pip check` step fails as the dependency versions are too old.
* Dependencies have been upgraded, and something is breaking. As long as brax "Live at Head" the risk of this I think is quite low. 

I can't think right now of other problems that can emerge. Anyhow, at any moment you are free to remove yourself from the list of mantainers.

traversaro, 2022-09-14:
> Hi @traversaro - thanks for looking into this! We could make `pytinyrenderer` an optional package - does conda have some notion of extras? `pytinyrenderer` is not used in our core training/visualization pipeline. Would that make things easier?

Unfortunatly conda does not have the notion of extra/optional deps. Anyhow, perhaps I was a bit too pessimistic on  `pytinyrenderer` , I may be able to package it in a few days.

traversaro, 2022-09-14:
I was a bit pessimistic, I was able to package pytinyrenderer with a couple patches: https://github.com/conda-forge/staged-recipes/pull/20437 .

erikfrey, 2022-09-15:
OK great!  I commented on the conda-forge PR.  Thanks for setting this up.

traversaro, 2022-10-06:
> I was a bit pessimistic, I was able to package pytinyrenderer with a couple patches: [conda-forge/staged-recipes#20437](https://github.com/conda-forge/staged-recipes/pull/20437) .

The PR was merged. We can proceed with the brax recipe.

traversaro, 2022-10-13:
The PR was merged, now `brax` is available from conda-forge . You can install it with `conda install -c conda-forge brax` or `mamba install -c conda-forge brax` , I guess we can close the issue.

erikfrey, 2022-10-13:
Amazing, thanks @traversaro !  We've updated the README with new installation instructions and will push that soon.  Closing this issue - cheers.

closed 2022-10-13 by erikfrey