google/brax #224: TPU training colab is not working for me

ViktorM, 2022-09-14:
TPU training colab stopped working for me after one of the updates in recent months. It stalls when comes to the training cell it shows that some work is going on but it never finishes, and no plots and trained policies are produced.

erwincoumans, 2023-03-20:
Hi, tried this TPU training colab today, and it fails with this error:
```

AttributeError                            Traceback (most recent call last)
[<ipython-input-3-9ce5fdb19302>](https://localhost:8080/#) in <module>
     14 
     15 try:
---> 16   import brax
     17 except ImportError:
     18   get_ipython().system('pip install git+https://github.com/google/brax.git@main')

2 frames
[/usr/local/lib/python3.9/dist-packages/brax/jumpy.py](https://localhost:8080/#) in <module>
    504 
    505 
--> 506 def where(condition: jax.typing.ArrayLike, x: jax.typing.ArrayLike,
    507           y: jax.typing.ArrayLike) -> ndarray:
    508   """Return elements chosen from `x` or `y` depending on `condition`."""

AttributeError: module 'jax' has no attribute 'typing'
```

btaba, 2023-03-20:
Hi @erwincoumans! I'm not able to reproduce the issue, which jax version are you using?

closed 2024-10-04 by btaba