google/brax #240: Wrong inertia and torque on Humanoid environment

imoneoi, 2022-10-11:
It seems that the inertia matrices of Humanoid environment are set arbitrarily (using identity matrices), and motor torques may be too large. I have strange behaviors in training like oscillating and flying around.

The bodies are identical as Gym MuJoCo Humanoid, and mass calculation are correct (assuming water density, 1000 kg / m^3).

BTW, what are the training parameters of caption video? The humanoid gait looks realistic.

imoneoi, 2022-10-12:
Update: I wrote code to infer the mass and inertia from geometry like MuJoCo does, and would like to PR this feature.

There is another question. Brax seems to use only the diagonal of inertia tensor. I think we can approximate by only taking the diagonal elements, or update the physics code to use a 3x3 tensor ?

cdfreeman-google, 2022-10-13:
Indeed, we typically use identity inertias in most of the environments, because the maximal-coordinate style methods are quite sensitive to the relative inertial scales between parts.  This has been qualitatively "good enough" for research on continuous control problems, because precise inertias only start becoming relevant once you want to do something like sim2real (and even then, you'd be surprised with how much you can get away with).

So, Bad News: this is working as intended, and we don't plan on fixing this in brax v1
Good News: brax v2 will be significantly more careful about this, and loads inertial data from mujoco xmls directly.  Additionally, the spring/pbd codepaths in v2 apply forces in the center-of-mass frame, which greatly simplifies the inertia math.

Long story short: thank you tremendously for looking into this, and I think you'll be much happier with how this is treated in v2 :D

erikfrey, 2022-10-13:
That said, big +1 to the quality of and care you showed in your PR.  I'd be overjoyed for you to contribute more in the future          :-)

imoneoi, 2022-10-14:
It's really sensitive. When I plugged in the correct inertia, the simulation become unstable (maybe I failed to tune the damping parameters). And using the identity inertia lead to different types of unrealistic behaviors.

Looking forward to brax v2. Are there any public available versions yet? I really need a fast JAX jittable physics simulation. I'm glad to contribute and improve it.

btaba, 2023-02-10:
Hi @imoneoi we posted a preview of brax v2 here:
https://github.com/google/brax/releases/tag/v0.1.0

where we implemented non-spherical inertias. We have a humanoid env here that can be used:

https://github.com/google/brax/blob/f0de0e56126b7cd7cdf941f0fc02f3480a02716a/brax/v2/envs/humanoid.py#L27

We also find that non-spherical inertias are very sensitive for spring and positional backends (as used in v1), and we plan to investigate how to make those more stable. Feel free to take a look! For now we set inertias to be spherical for spring/positional due to stability issues, but they are correct for the `generalized` backend. I'll close this issue for now

closed 2023-02-10 by btaba