google/brax #253: About documentation

jinseokbae, 2022-11-11:
Hi, 
Thanks for releasing amazing work.
I'm really interested in the differentiable physics simulator so I'm planning to use Brax on my recent research project.
However, I couldn't find any proper document explaining API of Brax.
Is there any publicly available document?
If not, when do you plan to release it?

btaba, 2022-11-12:
Hi @jinseokbae !
The best documentation we have now are the notebooks in the root directory here:  https://github.com/google/brax/tree/main/notebooks
We could certainly have better documentation! We plan on releasing a major version update of brax in the next couple of months first before adding more documentation. Thanks for considering brax!

evlabs-jventajas, 2023-03-15:
All those notebooks seem to be broken.

btaba, 2023-03-21:
@evlabs-jventajas can you try again and let us know if you face any issues? just pushed some fixes

btaba, 2024-10-04:
Closing due to inactivity