google/brax #254: Contact forces since PBD implementation

namheegordonkim, 2022-11-11:
I'm struggling to use fetch contact forces from `Info`. I used to use `info.contact.vel` with a particular body's index, but all of the entries of `info.contact.vel` and `info.contact.ang` are zeros now. Instead, there are `info.contact_pos`, `info.contact_normal`, and `info.contact_penetration`, but now the number of rows in these arrays are more than there are bodies in the environment, so I'm not sure how to index these. Any hints? Perhaps I'll revert to using `legacy_spring` for now...

namheegordonkim, 2022-11-11:
Sort of figured it out--still unsure why `info.contact.vel` and `info.contact.ang` are all zero, but here's what I found out.

`system.py` has `_get_contact_info()` which flattens contact-related arrays. The argument `contact` is a list of colliders (corresponding to possible collision pair according to collide_include), and depending on which pair (e.g. box-plane, capsule-capsule, etc.), the dimensions of these colliders will differ. For example, box-capsule one-way collision has 12 opportunities of contact (2 per side of box). If I have 4 collide_include statements for box-capsule collisions in my environment definition, I get 4*12 = 48 rows pertaining to them. It's hard to say which 48 of them among the many more rows; I think they're sorted in the order the pairs are processed.

I must say this is a very cryptic way to present contact information--presumably all the brax environments that used to include contact forces into observation stopped doing so because it's hard to parse the info. Any chance some quality-of-life refactors for this are coming soon?

btaba, 2022-11-11:
Hi @namheegordonkim !

`info.contact_*` are meant for the visualizer, which is why the order is a bit arbitrary; they aren't really meant to be human readable. It's odd that `info.contact.vel` and `info.contact.ang` are zero if you are seeing actual collisions, as these should be the contact impulses that are resolved in the integrator. You should try to use `info.contact.*`, please let us know if you see an issue there as this may point to an actual bug in the info.
We have been working on a major release of brax that will have a simpler API for getting contacts

namheegordonkim, 2022-11-14:
Okay, looks like `info.contact.*` values are not populated for frozen objects, e.g. ground, FWIW

btaba, 2022-12-20:
Hi @namheegordonkim frozen objects don't get updated, hence values not populated in `info.contact.*`, but they can still be visualized with `info.contact_*`. Sorry for the confusion here!

closed 2022-12-20 by btaba