google/brax #257: feat(composer): agent specific observations

cemlyn007, 2022-11-13:
Hey 👋

I am working in my fork on supporting [agent specific observations](https://github.com/google/brax/blob/8a3ea6065866f687cf657d4fc5c91d002a75f72f/brax/experimental/composer/agent_utils.py#L36) in the composer sub-package. Additionally I need this feature to support observation masking that is dependent on the current state of the environment.

Best approach I can see:
1. Each agent component could accept a dictionary of `observer_fns` that contain objects that implement the `Observer` interface
    a) Similar to how [`reward_fns`](https://github.com/google/brax/blob/b3e75f9f0a66c19a3d76f232218b5029f116f3dd/brax/experimental/composer/composer.py#L45) is done
    b) This could support dynamic masking
2. Pass this sort of information inside the `edge` parameter to `Composer`:
    a) Could be part of its responsibility[ "automatically create necessary edge information among 2+ components"](https://github.com/google/brax/blob/b3e75f9f0a66c19a3d76f232218b5029f116f3dd/brax/experimental/composer/composer.py#L47)
    b) But maybe edge information is only meant to care about [pairs of components](https://github.com/google/brax/blob/b3e75f9f0a66c19a3d76f232218b5029f116f3dd/brax/experimental/composer/composer.py#L207)

Let me know if this sounds cool with you guys, but otherwise I am going to make a start on this feature 😀

btaba, 2022-11-15:
Hi @cemlyn007 ! Afaik we won't be maintaining composer going forward unfortunately. That approach sounds reasonable though, although `edge` does seem to be designed around "pairs"

cemlyn007, 2022-11-15:
Does this mean the composer sub-package will be removed in your upcoming release?

Will your new release support composable environments with a different sub-package?

btaba, 2022-11-15:
@erikfrey for more info here on composer

erikfrey, 2022-12-15:
Yes, the `composer` package is in limbo - its original maintainer has moved on and we haven't seen enough usage to warrant pulling it out of `experimental` and officially supporting it.

BUT if you care deeply about this package and are willing to maintain it, please let us know!  I'll leave this issue open for a while as a call to action if anyone is interested.