google/brax #260: Colab Notebook "Brax Training with PyTorch on GPU" partially fails to run.

moritzf, 2022-11-16:
When you run the introductory Colab notebook [Brax Training with PyTorch on GPU](https://colab.research.google.com/github/google/brax/blob/main/notebooks/training_torch.ipynb), it fails to run the second to last code cell:

```
AttributeError                            Traceback (most recent call last)
[<ipython-input-4-4076a06b45b0>](https://l8hn377yq-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20221115-060110-RC00_488628369#) in <module>
     21   plt.show()
     22 
---> 23 train(progress_fn=progress)
     24 
     25 print(f'time to jit: {times[1] - times[0]}')

4 frames
[/usr/local/lib/python3.7/dist-packages/brax/io/torch.py](https://l8hn377yq-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20221115-060110-RC00_488628369#) in _devicearray_to_tensor(value, device)
     81                            device: Device = None) -> torch.Tensor:
     82   """Converts a Jax DeviceArray into PyTorch Tensor."""
---> 83   dpack = jax.dlpack.to_dlpack(value.astype("float32"))
     84   tensor = torch_dlpack.from_dlpack(dpack)
     85   if device:

AttributeError: module 'jax' has no attribute 'dlpack'
```

You can fix this simply by replacing `jax.dlpack` with `dlp` on line 52, and `import jax.dlpack as dlp` after line 21 in file `brax/io/torch.py`.

In summary:

**Expected**: Run all in [Brax Training with PyTorch on GPU](https://colab.research.google.com/github/google/brax/blob/main/notebooks/training_torch.ipynb) should run without error.
**Actual**:  `error module 'jax' has no attribute 'dlpack'` in second to last code block

I will submit a PR to fix this if you don't mind.

PSilvestre, 2022-11-30:
+1 I use brax in a PyTorch framework and am also affected by this issue. It is not limited to the Colab Notebook example. I think it might have been caused by a mismatch of JAX and brax versions.

ViktorM, 2022-12-05:
Any updates on this? The same error.

ozhanozen, 2022-12-13:
If you do not want to modify Brax files, as a temporary solution, you can add "import jax.dlpack" at the beginning of the notebook. Doing this solved the problem for me.

moritzf, 2022-12-14:
Fixed, thanks @erikfrey!

closed 2022-12-14 by moritzf