google/brax #269: `mujoco_convert.py` does not use the default values of MJCF

Theo-Cheynel, 2022-12-05:
I tried running mujoco_convert.py on a sample MJCF file provided on the dm_control repository ([this one](https://github.com/deepmind/dm_control/blob/4e1a35595124742015ae0c7a829e099a5aa100f5/dm_control/suite/ball_in_cup.xml), but others have similar issues).
I get errors such as "unsupported geom type : None", because the `type` attribute of the elements is not always defined in the MJCF, since it relies on the defaults defined by MuJoCo in their XML reference. For instance, [the documentation](https://mujoco.readthedocs.io/en/stable/XMLreference.html?highlight=geom#body-geom:~:text=type%3A%20%5Bplane%2C%20hfield%2C%20sphere%2C%20capsule%2C%20ellipsoid%2C%20cylinder%2C%20box%2C%20mesh%5D%2C%20%E2%80%9Csphere%E2%80%9D) seems to indicate that the default value for the attribute "type" is "sphere".

I asked over at dm_control to know whether the parse_xml function worked as intended or if they planned on adding the default values to it, and they said it was the expected behaviour. So I think brax should take those into account after the parse_xml function. 
What do you think ?

erikfrey, 2022-12-15:
We are soon releasing a new API for Brax that has native support for MJCF - leaving this issue open for now until you can try it, but it will support default values in the XML, along with a host of other fixes for MJCF compatability.

Stay tuned!

Theo-Cheynel, 2022-12-19:
Thanks ! Can't wait :rocket:

btaba, 2023-02-10:
We've posted about the update here:
https://github.com/google/brax/releases/tag/v0.1.0

Closing the issue for now, thanks for using brax @Theo-Cheynel !

closed 2023-02-10 by btaba