google/brax #271: "OpenAI Gym Wrapper + PyTorch" Colab notebook example crashes

Robokan, 2022-12-14:
"OpenAI Gym Wrapper + PyTorch" example  crashes every-time it is  run on a Google colab sever.

[image]

Full error log:

/usr/local/lib/python3.8/dist-packages/gym/envs/registration.py:440: UserWarning: WARN: The `registry.env_specs` property along with `EnvSpecTree` is deprecated. Please use `registry` directly as a dictionary instead.
  logger.warn(
/usr/local/lib/python3.8/dist-packages/gym/utils/passive_env_checker.py:31: UserWarning: WARN: A Box observation space has an unconventional shape (neither an image, nor a 1D vector). We recommend flattening the observation to have only a 1D vector or use a custom policy to properly process the data. Actual observation shape: (4096, 87)
  logger.warn(
/usr/local/lib/python3.8/dist-packages/gym/core.py:317: DeprecationWarning: WARN: Initializing wrapper in old step API which returns one bool instead of two. It is recommended to set `new_step_api=True` to use new step API. This will be the default behaviour in future.
  deprecation(
/usr/local/lib/python3.8/dist-packages/gym/wrappers/step_api_compatibility.py:39: DeprecationWarning: WARN: Initializing environment in old step API which returns one bool instead of two. It is recommended to set `new_step_api=True` to use new step API. This will be the default behaviour in future.
  deprecation(
/usr/local/lib/python3.8/dist-packages/gym/utils/passive_env_checker.py:174: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed a `seed` instead of using `Env.seed` for resetting the environment random number generator.
  logger.warn(
/usr/local/lib/python3.8/dist-packages/gym/utils/passive_env_checker.py:190: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed `return_info` to return information from the environment resetting.
  logger.warn(
/usr/local/lib/python3.8/dist-packages/gym/utils/passive_env_checker.py:195: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed `options` to allow the environment initialisation to be passed additional information.
  logger.warn(
/usr/local/lib/python3.8/dist-packages/gym/utils/passive_env_checker.py:137: UserWarning: WARN: The obs returned by the `reset()` method was expecting a numpy array, actual type: <class 'jaxlib.xla_extension.DeviceArray'>
  logger.warn(
/usr/local/lib/python3.8/dist-packages/gym/spaces/box.py:226: UserWarning: WARN: Casting input x to numpy array.
  logger.warn("Casting input x to numpy array.")
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
[<ipython-input-8-70894b27c973>](https://wswp6w4dopp-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20221212-060104-RC00_494669873#) in <module>
     10 
     11 # jit compile env.reset
---> 12 obs = gym_env.reset()
     13 
     14 # jit compile env.step

3 frames
[/usr/local/lib/python3.8/dist-packages/brax/io/torch.py](https://wswp6w4dopp-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20221212-060104-RC00_494669873#) in _devicearray_to_tensor(value, device)
     81                            device: Device = None) -> torch.Tensor:
     82   """Converts a Jax DeviceArray into PyTorch Tensor."""
---> 83   dpack = jax.dlpack.to_dlpack(value.astype("float32"))
     84   tensor = torch_dlpack.from_dlpack(dpack)
     85   if device:

AttributeError: module 'jax' has no attribute 'dlpack'

ozhanozen, 2022-12-14:
This looks like as a duplicate of #260.

moritzf, 2022-12-14:
You can try running the example[^1] again. Should be fixed now.

[^1]: https://colab.research.google.com/github/google/brax/blob/main/notebooks/environments.ipynb#scrollTo=JR1vEFf263z6

erikfrey, 2022-12-15:
Yep!  This is fixed as of version `0.0.16` and also working in the colabs now.

Robokan, 2022-12-15:
great this is now working for me with  0.0.16  in Colab and also fixes bug #270 allowing it to run on the m1 Mac

closed 2022-12-19 by btaba