google/brax #276: Cartesian Joint Position

andrewwarrington, 2022-12-19:
Hi Brax,

Is there a way to extract from an environment the cartesian coordinates of each of the joints in the model?  Getting the quaternion-based state out is easy enough, but is there an in-built way to convert/extract this to 3-D cartesian (x-y-z) coordinates that is aware of the models geometry. 

Thanks,
Andy

andrewwarrington, 2022-12-19:
I think this was referred to as `xpos` in the original MuJoCo [link](https://roboti.us/forum/index.php?threads/cartesian-coordinates-of-body.3376/#post-3654)

btaba, 2022-12-19:
Hi @andrewwarrington ! In the current version of brax, the physics state is stored in maximal coordinates, so you should be able to extract the 3-D cartesian coordinates from `QP` directly without any additional transformations (`pos` and `rot`). Hope that answers the question!