google/brax #289: How to compute sys.link.invweight?

gijskoning, 2023-01-13:
The V2 version Brax is just in time for my thesis and already helping me a lot! Thanks for all the work.
Currently I am working on training an agent to be robust against different parameters by adversarial examples. Changing the parameters of an brax environment is not yet trivial. I know v2 is still in beta, and figured out computing most of the parameters. 
However, not for invweight, do you know how to compute the invweight from the original System, or the relation with respect to the mass/size of the robot?

gijskoning, 2023-01-14:
Also, let me know if I can do anything for this repo with respect to system identification/ domain randomization!

btaba, 2023-01-26:
Hi @gijskoning thanks for using brax! What do you mean by invweight? Inverse inertia can be calculated like [so](https://github.com/google/brax/blob/main/brax/v2/spring/maximal.py#L58-L69) if that's what you're referring to

gijskoning, 2023-01-26:
I should have pointed to what I meant. Normally it is obtained by mujoco when loading in the xml file: 
https://github.com/google/brax/blob/6211f46bf908aa793bc502845afbf469f34e6bb3/brax/v2/io/mjcf.py#L327.
And used here: https://github.com/google/brax/blob/6211f46bf908aa793bc502845afbf469f34e6bb3/brax/v2/base.py#L246

Thanks for the link I have looked over this function. Hopefully, that represents the invweight. Will try it out tomorrow.

btaba, 2023-02-10:
Ah, sorry for misunderstanding, you can check out how it's computed here before it gets loaded into a brax System:
https://github.com/deepmind/mujoco/blob/main/src/engine/engine_setconst.c#L88