google/brax #294: pytinyrenderer is incompatible with Python 3.11

imoneoi, 2023-01-18:
Brax cannot be installed with pip for Python 3.11. The dependency pytinyrenderer is not compatible with the latest Python. pytinyrenderer seems like a course project and hasn't been updated for a while. Consider removing it or replacing it with OpenGL/JAX renderer? (see #108)

btaba, 2023-02-10:
Thanks @imoneoi for the report and @traversaro for adding pytinyrenderer to conda. Indeed, can confirm `pip install brax` does not work on Python 3.11. We've been planning to add a jax renderer but have not started  yet

traversaro, 2023-02-10:
For anyone interested, the only thing that makes pytinyrender incompatible with python 3.11 is its vendored old pybind11 version. If you install a recent pybind11 (for example via pip) and apply a patch to use an external pybind11 (see https://github.com/conda-forge/pytinyrenderer-feedstock/blob/main/recipe/devendor.patch) pytinyrender should build fine from source.

erwincoumans, 2023-03-20:
Can you try again? Latest pytinyrenderer 0.0.14 has updated pybind11 and a wheel for Python 3.11

btaba, 2024-10-04:
Closing since we dropped pytinyrenderer and use the MJ renderer instead