google/brax #320: Creates a brax system from a MuJoCo model (Custom environment)

amdee, 2023-03-14:
Hi, I am trying to recreate/create a custom environment using a MuJoCo model as shown in the [v2 READMe.md](https://github.com/google/brax/tree/main/brax/v2l) especially unitree a1 robot. My approach was as follows;

1. I created a *UnitreeA1* class to represent the robot (it includes path to mjcf file). It has the reset and step method.
2. Registered the class through the "__init__.py" file in the envs folder 
3. Updated the base.py and mjcf.py files to account for cylindrical shapes.
4. I am running into the _**"RuntimeError: unrecognized collider pair: (, )" error**_. I know what causes the _RuntimeError_. Do I need to implement methods for a contact of a cylinder with other types of geometric shapes? if so, can you please provide guidance how to go about that?
5. Is possible to provide guidance on how to recreate the unitree a1 example on the _[v2 README.md](https://github.com/google/brax/tree/main/brax/v2)_

Thanks for your help! I am super excited about Brax (especially v2). Once I can recreate the custom environments, I would like to contribute to the documentation.

btaba, 2023-03-30:
Hi @amdee thanks for using brax! If the runtime error is because you added a cylindrical shape to brax on your branch, then you would either have to [1] turn off collisions for that shape using `contype` `conaffinity` in the XML, or [2] implement the collision pair that is throwing the error. To implement the collision, here's the code reference for where that happens:
https://github.com/google/brax/blob/1d5fbd7cdc71708e8549e34a0aa485e7dc8a460c/brax/geometry/contact.py#L446

@erikfrey for the a1 example in the README

amdee, 2023-03-30:
@btaba Thanks for replying. I ended up approximating the cylindrical shape with a Capsule but at one point I ended wanted to implement the collision pair for the cylindrical shape. I will revisit the ideal, thanks for the info.