google/brax #323: `act_size` (v2) larger than the number of actuators

Theo-Cheynel, 2023-03-23:
Hi, 

I noticed something unexpected while playing with the humanoid environment in BRAXv2.
The MJCF file defines 17 actuators (linked to 17 joints + a free joint for the root of the character).
The `link_types` string of this system looks like `f2131312121`, which can be read as "free joint, 2 joints, 1 joint, etc". That is logical, as 2+1+3+1+3+1+2+1+2+1 = 17.
Now, the `act_size` of the system is expected to be 17 as well, since there are 17 motors associated to 17 hinge joints with 1DoF each. However, the `act_size` of this system is not 17 but 35, counting 2DoFs for each joint in a two-joints configuration, and 3DoFs for each joint in a three-joints configuration. Indeed, 2+2 + 1 + 3+3+3 + 1 + 3+3+3 + 1 + 2+2 + 1 + 2+2 + 1 = 35.

I struggle to see if this is the expected behaviour, it seems to me that the act_size should be the sum of the number of DoFs of each joint, in our case, 17.
Can someone explain if my understanding is correct ?

Thanks in advance

btaba, 2023-03-30:
Hi @Theo-Cheynel  thanks for pointing this out, should be fixed now!

Theo-Cheynel, 2023-03-30:
Thanks ! I'll have a look :smile: 

In the meatime, I added pseudo bodies in between all my joints, to convert 3DoF joints to three 1 DoF joints :laughing:

btaba, 2023-03-30:
Ah got it, that is indeed annoying to have to do, sorry for the delay!