google/brax #329: unstable at smaller inertia values

Flogg0, 2023-04-05:
Hi,
I am currently working on a project where I want to replicate this robot in brax
[image]
It was originally simulated in pybullet and described as an urdf file.

I rebuilt the robot as a mujoco file similar to the examples in envs and enabled "inertiafromgeom". When visualizing the simulation in the first few steps everything seems normal then the robot quickly spirales out of control and then does not exist anymore.

I did further testing where i spawned the 3 parts of the robot as free moving objects and let them fall to the floor. They again fall normally towards the floor until they contact it and also cease to exist. Which makes me believe that something odd is happening when calculating with the bodies inertias since no torque should be applied to the bodies before they hit the floor. I checked against the reference inertia values from the urdf file and the inertias computed from the geometry were quite similar.  Iterations up to 1e5 did not fix the issue with dt= 0.01 and generalized backend.
Did i do a bad setup or is this a bug?

Further Information:
Position of objects before they hit the floor:
[image]

After they hit the floor:
[image]

I set up a repo where you could replicate this problem  with the robot and with the parts falling to the floor:
https://github.com/Flogg0/inertia_demo

btaba, 2023-04-07:
Thanks @Flogg0 for creating the issue and for the detailed explanation (and nice github repo). I was able to get stable simulation with:

```xml
    <numeric data="0" name="matrix_inv_iterations"/>
    <numeric data="50" name="solver_maxls"/>
```

using an exact mass matrix inv solver rather than the approximate one, and tweaking the projected gradient solver param. You might need some other tweaking (damping, lowering timestep -> you can set n_frames to skip the added frames, etc.). Hope that helps for now

Flogg0, 2023-04-09:
Thank you very much, this seemed to fix my problem

closed 2023-04-10 by btaba