google/brax #333: `position` servo parameters ?

Theo-Cheynel, 2023-04-11:
Hi,

I was wondering whether the `<position>` actuator was supported in v2, and whether we could change its gain parameters (in MJCF files, [the gains can be specified](https://mujoco.readthedocs.io/en/stable/XMLreference.html#actuator-motor:~:text=a%20position%20servo.-,The%20underlying%20general%20attributes%20are%20set%20as%20follows%3A,-Attribute)).

Thanks

btaba, 2023-04-11:
Yes it's supported. We don't use gain, just gear, in a somewhat simplified version of the position actuator (still subject to change)
the torque for a position actuator is $gear * (ctrl - q)$

Theo-Cheynel, 2023-04-11:
I see, thank you very much ! So there is only a proportional term, and no derivative term (w.r.t. $\dot{q}$) ?

btaba, 2023-04-11:
the derivative term is added via damping of the joint