google/brax #349: brax very slow?

ikamensh, 2023-04-28:
Hello, I've been testing brax and for some reason it's extremely slow for me, both on CPU and GPU. I've tried halfcheetah and reacher envs, both are same order of magnitude slow. Here is code that reproduces the issue:

```python
import time
import random

import torch
from brax import envs
from brax.io import torch as torch_interface
import jax

device = "cuda" if torch.cuda.is_available() else "cpu"
print(f"Using device: {device}")

batch_size = 1
print(f"batch size is {batch_size}")
e = envs.create("halfcheetah", batch_size=batch_size, episode_length=None)  # type:ignore[arg-type]
seed = random.randint(0, 10_000)
jseed = jax.random.PRNGKey(seed)
t = time.time()
s = e.reset(jseed)
print(f"{time.time() - t:.3f} sec to RESET")
for i in range(5):
    t = time.time()
    action = torch.rand([batch_size, e.action_size], device=device, dtype=torch.float32)
    jax_action = torch_interface.torch_to_jax(action)
    s = e.step(state=s, action=jax_action)
    print(f"{time.time() - t:.3f} sec to STEP")
```

Results:
Using device: cpu
batch size is 1
3.549 sec to RESET
17.961 sec to STEP
15.613 sec to STEP
14.953 sec to STEP
15.006 sec to STEP
14.901 sec to STEP
-> more than a minute to run 5 steps on halfcheetah?..

GPU:
Using device: cuda
batch size is 1
12.481 sec to RESET
29.761 sec to STEP
21.937 sec to STEP
21.978 sec to STEP
21.954 sec to STEP
21.945 sec to STEP

-> even worse on V100 GPU.

CPU env:
```
~/b/rlcore (main|●1✚1) $ pip freeze | grep ax                                                                                    (rlcore) 12:45:33
brax @ git+https://github.com/google/brax.git@ac3b374500e3c41c0818c48839f6ce66db27965c
flax==0.6.7
jax==0.4.6
jax-jumpy==0.2.0
jaxlib==0.4.6
jaxopt==0.6
```

GPU env:
```
w019. ~/b/rlcore pip freeze | grep ax                         (rlcore310)
brax==0.1.2
flax==0.6.9
jax==0.4.8
jax-jumpy==1.0.0
jaxlib @ file:///home/kai7rng/jaxlib-0.4.7%2Bcuda11.cudnn86-cp310-cp310-manylinux2014_x86_64.whl
jaxopt==0.6
```

ikamensh, 2023-04-28:
if that helps, I've profiled it.

On CPU:
```
   Ordered by: cumulative time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
 11501/20    0.017    0.000  148.513    7.426 traceback_util.py:162(reraise_with_filtered_traceback)
  2056/74    0.023    0.000  148.275    2.004 linear_util.py:151(call_wrapped)
12224/7151    0.075    0.000  142.707    0.020 profiler.py:311(wrapper)
124921/28068    0.236    0.000  141.202    0.005 core.py:362(bind_with_trace)
32937/15713    0.232    0.000  136.464    0.009 batching.py:344(process_primitive)
        5    0.017    0.003  135.135   27.027 half_cheetah.py:178(step)
24287/10849    0.062    0.000  134.719    0.012 core.py:2571(bind)
5154/5145    0.011    0.000  124.096    0.024 core.py:806(process_primitive)
     3603    0.011    0.000  118.246    0.033 dispatch.py:120(apply_primitive)
     4104    0.018    0.000  118.072    0.029 util.py:249(wrapper)
      312    0.094    0.000  117.840    0.378 dispatch.py:186(xla_primitive_callable)
      648    0.025    0.000  103.126    0.159 pxla.py:3547(from_hlo)
      648    0.014    0.000  102.564    0.158 dispatch.py:1050(compile_or_get_cached)
      648  102.479    0.158  102.479    0.158 dispatch.py:1026(backend_compile)
1608/1107    0.023    0.000   25.136    0.023 partial_eval.py:2034(trace_to_subjaxpr_dynamic)
      692    0.048    0.000   19.989    0.029 pxla.py:2898(lower_sharding_computation)
      648    0.076    0.000   18.776    0.029 mlir.py:657(lower_jaxpr_to_module)
100634/71904    0.117    0.000   18.155    0.000 core.py:357(bind)
 3341/648    0.224    0.000   18.098    0.028 mlir.py:850(lower_jaxpr_to_fun)
 3766/648    0.983    0.000   17.788    0.027 mlir.py:1105(jaxpr_subcomp)
10979/10371    0.035    0.000   14.143    0.001 pjit.py:233(cache_miss)
10979/10371    0.067    0.000   14.099    0.001 pjit.py:178(_python_pjit_helper)
8831/8610    0.405    0.000   12.610    0.001 core.py:410(eval_jaxpr)
     29/9    0.019    0.001   12.186    1.354 loops.py:1442(_while_lowering)
10380/8876    0.087    0.000    9.429    0.001 pjit.py:1507(_pjit_batcher)
86723/76582    0.120    0.000    9.178    0.000 partial_eval.py:1778(process_primitive)
     7838    0.063    0.000    8.840    0.001 lax_numpy.py:5107(deferring_binary_op)
791966/791626    1.840    0.000    8.525    0.000 util.py:74(safe_map)
    78623    0.447    0.000    8.097    0.000 partial_eval.py:1783(default_process_primitive)
2693/1224    0.076    0.000    7.015    0.006 pjit.py:1467(_pjit_lowering)
       90    0.015    0.000    6.350    0.071 math.py:73(quat_mul)
     4274    0.035    0.000    5.934    0.001 lax_numpy.py:3897(_rewriting_take)
     1541    0.013    0.000    5.820    0.004 pjit.py:1273(_pjit_call_impl)
    74509    0.097    0.000    5.493    0.000 core.py:395(abstract_eval_)
      258    0.015    0.000    5.311    0.021 math.py:25(rotate)
27387/21770    0.149    0.000    5.205    0.000 tree_util.py:173(tree_map)
59629/47473    0.027    0.000    5.015    0.000 tree_util.py:209(<genexpr>)
   435604    0.408    0.000    3.942    0.000 core.py:1482(__init__)
     3994    0.018    0.000    3.843    0.001 slicing.py:683(dynamic_slice_in_dim)
    54605    0.344    0.000    3.831    0.000 utils.py:55(standard_abstract_eval)
   477400    0.651    0.000    3.746    0.000 core.py:2011(canonicalize_shape)
389371/389143    0.390    0.000    3.676    0.000 core.py:691(__getattr__)
   215543    0.332    0.000    3.519    0.000 core.py:449(full_raise)
     3994    0.014    0.000    3.509    0.001 slicing.py:61(dynamic_slice)
    57224    0.073    0.000    3.337    0.000 core.py:2475(used_axis_names)
145275/131077    0.253    0.000    3.287    0.000 batching.py:267(aval)
    43713    0.367    0.000    3.259    0.000 mlir.py:322(_source_info_to_location)
130160/57224    0.194    0.000    3.238    0.000 core.py:2480(subst_axis_names)
   674360    0.635    0.000    3.112    0.000 core.py:2000(_canonicalize_dimension)
39882/35021    0.181    0.000    2.944    0.000 core.py:2550(subst_axis_names_jaxpr)
6951128/6948958    1.569    0.000    2.736    0.000 {built-in method builtins.isinstance}
592032/585970    0.216    0.000    2.727    0.000 {built-in method builtins.next}
10979/10470    0.054    0.000    2.574    0.000 api.py:435(infer_params)
10979/10470    0.295    0.000    2.513    0.000 pjit.py:423(common_infer_params)
1267/1170    0.033    0.000    2.398    0.002 core.py:2544(used_axis_names_jaxpr)
    12139    0.032    0.000    2.368    0.000 pjit.py:1446(pjit_staging_rule)
1267/1170    0.027    0.000    2.255    0.002 core.py:2537(<listcomp>)
72936/71544    0.135    0.000    2.228    0.000 core.py:2519(subst_axis_names_eqn)
9745/8899    0.031    0.000    2.212    0.000 lax.py:813(broadcast_in_dim)
183352/154688    0.087    0.000    2.199    0.000 core.py:1344(get_aval)
454033/454023    0.254    0.000    2.130    0.000 {built-in method builtins.hasattr}
    19185    0.027    0.000    2.128    0.000 lax.py:2877(_broadcast_in_dim_staging_rule)
  2148429    1.511    0.000    2.104    0.000 config.py:240(get_state)
[... 282 lines cut ...]
    94510    0.024    0.000    0.029    0.000 dtypes.py:447(_jax_type)
     7612    0.014    0.000    0.028    0.000 dispatch.py:257(is_single_device_sharding)
     1979    0.020    0.000    0.027    0.000 inspect.py:3050(_bind)
     3112    0.011    0.000    0.025    0.000 core.py:1890(_dim_handler_and_canonical)
     6000    0.011    0.000    0.025    0.000 __init__.py:430(_replace)
[... cut ...]
   105595    0.016    0.000    0.016    0.000 dtypes.py:637(<genexpr>)
      821    0.011    0.000    0.015    0.000 function_base.py:1319(diff)
    39649    0.015    0.000    0.015    0.000 tree_util.py:642(<listcomp>)
    70014    0.014    0.000    0.014    0.000 {built-in method builtins.setattr}
     7391    0.013    0.000    0.013    0.000 util.py:418(__eq__)
    64756    0.011    0.000    0.011    0.000 core.py:1375(<lambda>)
      818    0.010    0.000    0.010    0.000 {built-in method jaxlib.mlir._mlir_libs._mlir.ir.file}
```

btaba, 2023-04-28:
Hi @ikamensh , can you jax.jit the e.step and e.reset functions?

For example:

```python
jit_step =  jax.jit(e.step)
```

then use `jit_step` instead of `e.step`

erikfrey, 2023-04-28:
Running without jit is useful for stepwise debugging, but for anything where perf matters, you should run your functions jitted.  Two other things to keep in mind:

* For optimal perf you will want to run batched environments.  Even for testing it's often useful to run batched environments to better understand the distrubution of a policy.
* Using the [torch and gym env wrappers](https://github.com/google/brax/tree/main/brax/envs/wrappers) (instead of manually converting between torch and jax tensors as you did in your code) may make your workflow easier.  It also automatically handles jitting for you.

Please see our [torch colab ](https://colab.research.google.com/github/google/brax/blob/main/notebooks/training_torch.ipynb) for an example of putting these principles together.  Properly configured, you can get a half million steps per second on a GPU with Torch and Brax.

oarriaga, 2023-04-28:
Hi @ikamensh I made a small example with new brax API in `generalized` coordinates with `jit` and it works with

```python
import time
import jax
from brax import envs

env = envs.create(env_name='halfcheetah', backend='generalized')
reset = jax.jit(env.reset)
step = jax.jit(env.step)
seed = jax.random.PRNGKey(777)
reset_seed, state_seed = jax.random.split(seed)
t = time.time()
state = reset(reset_seed)
print(f"{time.time() - t:.3f} sec to RESET")
for i in range(5):
    t = time.time()
    state_seed, seed = jax.random.split(state_seed)
    jax_action = jax.random.normal(seed, (env.action_size,))
    state = step(state=state, action=jax_action)
    print(f"{time.time() - t:.3f} sec to STEP")

```

btaba, 2023-05-02:
Closing the issue for now, but please re-open if you have any issues with jitted speed

closed 2023-05-02 by btaba