google/brax #350: Why Were the Tendons on `Humanoid`'s models removed (hip knee)?

Kallinteris-Andreas, 2023-05-04:
The original gymnasium Humanoid had 2 tendons:
https://github.com/Farama-Foundation/Gymnasium/blob/2feeb31378942ff3373915a4fe31faffb0d2b26c/gymnasium/envs/mujoco/assets/humanoid.xml#L91

Which are not present in the `brax` model

https://github.com/google/brax/blob/c2cd14cf762242d63aeec106d955390c8e14d582/brax/envs/assets/humanoid.xml

Is there a reason for that?

btaba, 2023-05-04:
Hi @Kallinteris-Andreas , thanks for using brax! As of now brax does not support tendons

Kallinteris-Andreas, 2023-05-04:
Is there an approximate ETA for tendon support? Is it on the roadmap?

btaba, 2023-05-04:
Hi @Kallinteris-Andreas it's not really on the roadmap or high priority, but if you'd like to take a stab please feel free!

closed 2023-05-04 by Kallinteris-Andreas