google/brax #358: Unsupported Hinge Joints/Actuators

tomalner18, 2023-05-11:
Hi, I'm loading a system that contains the following agent, exported to XML from MuJoCo:

```
<mujoco>
	<asset>
		<texture name='smiley' file="smiley.png" gridsize="3 4" gridlayout=".U..LFRB.D.."/>
		<material name="matSmiley" texture="smiley" specular="1" shininess=".3" reflectance="0"/>
	</asset>
	<option timestep="0.005"/>
	<worldbody>
		<body name="particle" pos="0.15 0.15 0.15">
			<joint axis="1 0 0" damping="10" name="tx" pos="0 0 0" type="slide"></joint>
			<joint axis="0 1 0" damping="10" name="ty" pos="0 0 0" type="slide"></joint>
			<joint axis="0 0 1" damping="10000" name="tz" pos="0 0 0" type="slide"></joint>
			<joint axis="0 0 1" damping="10" name="rz" pos="0 0 0" type="hinge"></joint>
			<geom name="agent" material="matSmiley" mass="1" pos="0 0 0" rgba="1 0 0 1" size="0.15" type="sphere" euler="1.57 0 0"></geom>
			<geom name="arm1" mass="0" pos="-0.13 -0.075 -0.05" rgba="1 0 0 1" size="0.05" type="sphere" euler="1.57 0 0"></geom>
			<geom name="arm2" mass="0" pos="0.13 -0.075 -0.05" rgba="1 0 0 1" size="0.05" type="sphere" euler="1.57 0 0"></geom>
		</body>
		<body name="annotation:outer_bound" pos="0.15 0.15 0.15">
			<geom conaffinity="0" contype="0" mass="0" pos="0 0 0" rgba="0.417 0.7203 0.0001 0.1" size="0.15 0.15 0.15" type="box"></geom>
		</body>
	</worldbody>
	<actuator>
		<motor name="motortx" gear="100" joint="tx"></motor>
		<motor name="motorty" gear="100" joint="ty"></motor>
		<motor name="motorrz" gear="100" joint="rz"></motor>
	</actuator>
</mujoco>
```

The Mjcf parser throws  `NotImplementedError: Link types {'4'} are not supported for actuators.`
I'm pretty sure the issue is with parsing the hinge joint and associated motor; three translational joints is accepted, but two translational and the rotational fails. Is this just a system that isn't implemented yet, or can I make any changes to my agent to allow rotation in that dimension?
Thanks!

tomalner18, 2023-05-15:
Also, I'm unsure how I could create a fixed body using the generalised pipeline (mimic the frozen behaviour from v1). Creating a geom without a body works with the other pipelines, but fails the constraint tests. In MuJoCo it seems possible by creating bodies without joints, is equivalent functionality currently implemented in Brax? Thanks for any help

erikfrey, 2023-05-18:
Hi Tom - your config looks like it contains **three** translational and one rotational?  Right now, Brax supports free (6-dof), 1-dof, 2-dof, and 3-dof joints.  The error you got was not very helpful though!  I'll make a note of that, thank you.

Yes indeed you can create geoms in v2 that are not attached to links (bodies).  See for example the ground plane in our [gym environment assets](https://github.com/google/brax/blob/main/brax/envs/assets/ant.xml).

Not quite sure I follow what you mean by "fails the constraint tests"?

tomalner18, 2023-05-23:
Hi Erik, thanks for helping out! Sorry yes, the example I gave did have **three** translational joints and one rotational. Whilst I don't get an error with a file like this(2 + 1)?:
```
<mujoco>
	<option timestep="0.005"/>
	<worldbody>
		<geom name="floor0" pos="3 3 0" size="3 3 1" type="plane" condim="3"></geom>
		<body name="floor0" pos="3 3 0"></body>
		<body pos="0.15 0.15 0.15">
			<joint axis="1 0 0" damping="0.1" name="slide0" pos="0 0 0" type="slide"></joint>
			<joint axis="0 1 0" damping="0.1" name="slide1" pos="0 0 0" type="slide"></joint>
			<joint axis="0 0 1" damping="0.1" name="hinge2" pos="0 0 0" type="hinge"></joint>
			<geom name="agent" mass="1" pos="0 0 0" rgba="1 0 0 1" size="0.15" type="sphere" euler="1.57 0 0"></geom>
		</body>
		<body name="annotation:outer_bound" pos="0.15 0.15 0.15">
			<geom conaffinity="0" contype="0" mass="0" pos="0 0 0" rgba="0.417 0.7203 0.0001 0.0" size="0.0001 0.0001 0.0001" type="box"></geom>
		</body>
	</worldbody>
	<actuator>
		<motor gear="100" joint="slide0"></motor>
		<motor gear="100" joint="slide1"></motor>
	</actuator>
</mujoco>
```
I do get a link related error though when I add bodies with joints to that system `(IndexError: list index out of range)`. Sorry if I'm missing something obvious:
```
<mujoco>
	<option timestep="0.005"/>
	<worldbody>
		<geom name="floor0" pos="3 3 0" size="3 3 1" type="plane" condim="3"></geom>
		<body name="floor0" pos="3 3 0"></body>
		<body pos="0.15 0.15 0.15">
			<joint axis="1 0 0" damping="0.1" name="slide0" pos="0 0 0" type="slide"></joint>
			<joint axis="0 1 0" damping="0.1" name="slide1" pos="0 0 0" type="slide"></joint>
			<joint axis="0 0 1" damping="0.1" name="hinge2" pos="0 0 0" type="hinge"></joint>
			<geom name="agent" mass="1" pos="0 0 0" rgba="1 0 0 1" size="0.15" type="sphere" euler="1.57 0 0"></geom>
		</body>
		<body name="annotation:outer_bound" pos="0.15 0.15 0.15">
			<geom conaffinity="0" contype="0" mass="0" pos="0 0 0" rgba="0.417 0.7203 0.0001 0.0" size="0.0001 0.0001 0.0001" type="box"></geom>
		</body>
		<body name="moveable-box0" pos="0.25 0.25 0.25">
			<joint type="free" name="moveable-box0_free" damping="0.01" pos="0 0 0"></joint>
			<geom size="0.25 0.25 0.25" type="box" condim="3" name="moveable-box0" rgba="1 0.5 0.8 1" mass="1" friction="0.2"></geom>
		</body>
		<body name="moveable-box1" pos="0.25 0.25 0.25">
			<joint type="free" name="moveable-box1_free" damping="0.01" pos="0 0 0"></joint>
			<geom size="0.25 0.25 0.25" type="box" condim="3" name="moveable-box1" rgba="1 0.5 0.8 1" mass="1" friction="0.2"></geom>
		</body>
	</worldbody>
	<actuator>
		<motor gear="100" joint="slide0"></motor>
		<motor gear="100" joint="slide1"></motor>
	</actuator>
</mujoco>
```
 _By constraint tests, I think I meant an assertion in a jacobian function in constraint.py._ Also, if you don't mind answering, I'm quite interested in why 4-dof isn't supported in Brax. Doesn't the generalised pipeline use the same physics as MuJoCo, which supports 4-dof? Thanks!

btaba, 2023-05-31:
Hi @tomalner18 ! I ran the second config above using the following code:

```python
sys = mjcf.loads(config)
s = jax.jit(g_pipeline.init)(sys, sys.init_q, jp.zeros(sys.qd_size()))
s = jax.jit(g_pipeline.step)(sys, s, jp.zeros(sys.act_size()))
```

and don't get an error, How can we reproduce the IndexError?

RE 4-dofs not supported, this was a design decision so that we can keep supporting spring and positional backends, which operate in maximal coordinates; joint constraints are more or less handled in a case-by-case basis.

tomalner18, 2023-05-31:
Hey @btaba! Embarrassingly I've just found that error was caused by some of my old code incorrectly parsing the xml before loading the system and adding unnecessary slide joints 😅 . Sorry for the false alarm, nothing wrong on your end! Thank you for the willingness to help out and the explanation.

closed 2023-05-31 by tomalner18