google/brax #361: Question about the docstrings

OnatV, 2023-05-22:
Hi, I am trying to understand the codebase and I got confused by some of the explanations. In **base.System** class, you define a property "link" which is explained as _link:(num_link,) the links in the system_.  However, in the implementation it corresponds to a single **base.Link** instance which is not subscriptable. There are also similar properties such as "link_names", which are explained in the same way but are plain lists. I understand "link_types" correspond to joint types and "link_names" correspond to names of the different kinematic chains. I couldn't make the exact connection between these fields and the **base.Link** class. I would greatly appreciate if you can give me an explanation on the topic.

erikfrey, 2023-05-22:
Hello!

This notation (such as `(num_link,)` describes the size of the topmost dimension of the arrays in this pytree.  This is sometimes called the batch dimension.  Batch dimensions add onto the existing shape of members of the pytree.  So for example, if `transform.pos` typically has a shape of `(3,)`, then the shape of `System.link.transform.pos` ought to have shape `(num_link, 3)`

Does that answer your question?

btaba, 2023-05-31:
Hi @OnatV , yep all attributes of the Link class should have a first dimension of num_link

closed 2023-06-09 by btaba