google/brax #370: HTML visualizer follow target and freeze angle features not working in Brax v2

imoneoi, 2023-06-08:


btaba, 2023-06-09:
Hey @imoneoi , would you have a simple example that we can use to reproduce the issue you're seeing, along with a description of the expected behavior and failure mode?

imoneoi, 2023-06-10:
Of course, this is a humanoid video trained with Brax v2 and exported via `brax.io.html.save`. As you can see in this HTML, the camera cannot follow the humanoid and freezes the angle.

In videos exported with Brax v1, the camera can follow the humanoid and use a fixed angle.

[1000.zip](https://github.com/google/brax/files/11712378/1000.zip)
[1000_v1.zip](https://github.com/google/brax/files/11712403/1000_v1.zip)

arxaqapi, 2023-06-13:
Noticed the same problem for the `follow target` option with Brax v2.
Followed the example shown [here](https://github.com/google/brax/blob/main/notebooks/training.ipynb).

My pseudo-code looks like this:

```python

env, pipeline_states = train(...)

html_string = html.render(
    env.sys.replace(dt=env.dt),
    pipeline_states,
    height="100vh", colab=False
)

with open(f"visualization.html", "w") as f:
    f.writelines(html_string)
```

erikfrey, 2023-06-14:
Hello!  This is fixed in ac2b94bad71658a2cc194518c6585c11117663fe and will be generally available in the next release.

closed 2024-10-04 by btaba