google/brax #383: Wrong behavior with `genearlized` and instability with `positional` or `spring`

UltronAI, 2023-08-04:
Hi,

I recently conducted experiments using the provided [Brax Basics Colab](https://colab.research.google.com/github/google/brax/blob/main/notebooks/basics.ipynb) and trained the built-in APG algorithm to investigate the impact of different physics backends. I observed some peculiar behaviors during these experiments, which I'd like to share with you.

1. Tossing the ball:
In this task, the objective is for the ball to bounce up after hitting the ground. The default "positional" backend achieved the desired behavior and looked good. However, when I switched to the "generalized" backend, the ball rolled along the ground without bouncing, which is not physically accurate. To help visualize the differences between the "positional", "generalized" and "spring" physics backends, I plotted their respective behaviors using the colors R, G, and B for easy comparison.
Here's the visualization:

[image]

2. Swinging the pendulum

In this task, the "generalized" backend performed well, providing the expected results. However, the "positional" backend collapsed for all step sizes, and the "spring" backend exhibited instability when the step size increases.

[image]

[image]

3. Pointing the pendulum

In this task, only `generalized` worked well. Both `positional` and `spring` collapsed.

amine789, 2023-08-05:
Hi i am trying to learn the elasticity parameters in the positional based dynamics by generating two scenarios, one simulation scenario with the real elasticity parameters and the other one with random elasticity, then compute the loss wrt positions but i am getting elasticity gradient wrt to position equal to zero, did you get something similar?

UltronAI, 2023-08-06:
Hi, @amine789 
While I haven't directly experimented with elasticity parameters, I've discovered some relevant insights in [this paper](https://openreview.net/forum?id=KIl0LZ9tJex). See the data presented in the final row of Table 1. The authors provided a reasonable explanation in the corresponding section. Hope it's helpful for you!

btaba, 2023-08-31:
Hi @UltronAI , thanks for the insightful experiments using brax!

[1] Tossing the ball

the restitution params between spring/positional and generalized are quite different. For spring/positional, they are controlled by the `elasticity` param (https://github.com/google/brax/blob/main/brax/io/mjcf.py#L182) and for generalized we use the contact model from MuJoCo. The restitution for generalized is controlled by the solver params, see https://mujoco.readthedocs.io/en/latest/modeling.html#restitution

Other params that may affect restitution in positional are the physics timestep and collide_scale, amongst others.

[2] Swinging the pendulum

I would recommend tuning parameters to maintain stability in positionl//spring. Joint constraints are implicitly maintained in generalized since it uses Featherstone's. For positional/spring, joint constraints are resolved at every time step and are likely  to be more unstable, and need to be tuned (ang/linear damping etc.).

It isn't clear if you are training a policy for swinging, and are making stability conclusions based on the final policy or the physics?

[4] training

Thanks for the findings! The contact impulse are backed out through the constraint solver in generalized, and it is thus not too surprising that the gradients may be less useful (autograd through a constraint solve) compared to the simple impulse-based contact updates in spring/positional.  

For further reference on the contact model for generalized, see https://mujoco.readthedocs.io/en/latest/computation.html#contact