google/brax #384: Gradient of elasticity wrt to position

amine789, 2023-08-05:
Hi Guys, Thank you for this work,

i am trying to learn the contact parameters from simulation data but the problem i have, is that i have a real simulation data and pred simulation data with random elasticity value(everything the same except elasticity)? the goal i want to learn elasticity value by computing the loss between real position and pred position then compute the gradient of elasticity and update the elasticity parameter value? the problem when i compute the gradient wrt to velocity i get a non-zero gradient but when i compute it wrt to position i get a zero gradient? any one has an idea why this is the case?

btaba, 2023-09-07:
Hi @amine789 , is the gradient of position computed wrt the entire trajectory? Also which physics backend are you using? Can you provide a reproducible script?
For the sake of simplicity / sanity check, I would first use a black-box optimizer to make sure you can optimize position for elasticity

amine789, 2023-09-08:
the gradients are working fine, it was a bug in the code

closed 2023-09-08 by amine789