google/brax #385: Actuator only has 6 fields ?

baixianger, 2023-08-12:
After parsing to Brax, I found the actuators in my model had lost some key parameters `force_range`. 

[image]

And I also looked up the source code from `brax.base.Actuator class` https://github.com/google/brax/blob/ac2b94bad71658a2cc194518c6585c11117663fe/brax/base.py#L449
Noticeable, it has 8 fields,   q_id, qd_id, ctrl_range, force_range,  gain, gear, bias_q, bias_qd.  And I am sure in my mjcf file I give the correct force range to each actuator.

I would appreciate any help from you.

baixianger, 2023-08-12:
Here attachs my actuators' setting.

```
    <default class="panda">
      <material specular="0.5" shininess="0.25"/>
      <joint armature="0.1" damping="1" axis="0 0 1" range="-2.8973 2.8973"/>
      <general dyntype="none" biastype="affine" ctrlrange="-2.8973 2.8973" forcerange="-87 87"/>
      <default class="finger">
        <joint axis="0 1 0" type="slide" range="0 0.04"/>
      </default>
      <default class="dummy">
        <joint axis="0 1 0" type="slide" range="0 1e8"/>
      </default>
      <default class="collision">
        <geom type="mesh" group="3"/>
      </default>
      <site size="0.001" rgba="0.5 0.5 0.5 0.3" group="4"/>
    </default>
<worldbody> ... </worldbody>
  <actuator>
    <!-- 作用在joint上的forcerange指的是力矩 -->
    <general class="panda" name="actuator1" joint="joint1" gainprm="4500" biasprm="0 -4500 -450"/>
    <general class="panda" name="actuator2" joint="joint2" gainprm="4500" biasprm="0 -4500 -450" ctrlrange="-1.7628 1.7628"/>
    <general class="panda" name="actuator3" joint="joint3" gainprm="3500" biasprm="0 -3500 -350"/>
    <general class="panda" name="actuator4" joint="joint4" gainprm="3500" biasprm="0 -3500 -350" ctrlrange="-3.0718 -0.0698"/>
    <general class="panda" name="actuator5" joint="joint5" gainprm="2000" biasprm="0 -2000 -200" forcerange="-12 12"/>
    <general class="panda" name="actuator6" joint="joint6" gainprm="2000" biasprm="0 -2000 -200" forcerange="-12 12" ctrlrange="-0.0175 3.7525"/>
    <general class="panda" name="actuator7" joint="joint7" gainprm="2000" biasprm="0 -2000 -200" forcerange="-12 12"/>
  </actuator>
```

baixianger, 2023-08-12:
My brax source code：
[image]

Source code from git
[image]

Did you ever publish a version with different Actuator dataclass settings? 😢

baixianger, 2023-08-12:
Has anyone else met this problem? or I had installed a fake Brax? 😭

btaba, 2023-10-15:
Hi @baixianger , sorry for the late reply. It looks like `force_range` was added several months after the `Actuator` class, you may have installed two different versions

closed 2023-10-15 by btaba