google/brax #391: Strange physics with humanoid model floor contact

areiner222, 2023-09-05:
I'm seeing some strange motion when simulating a humanoid system. I've enabled only body-floor contact and the initial foot contact and resulting motion under gravity alone doesn't appear natural.
The model has some complex geometries and it's tricky to encapsulate the code here for full repro, so I've attached a serialized system and example trajectory.

Let me know if there is any other way I could help demonstrate the issue.
[humanoid.json.zip](https://github.com/google/brax/files/12517478/humanoid.json.zip)

btaba, 2023-10-15:
Hi @areiner222 , can you link to a gif or video? I suspect this may be a similar issue to this: https://github.com/google/brax/discussions/396#discussioncomment-7287181

Which physics backend are you using?

areiner222, 2023-10-16:
Hi @btaba , I think this was an issue on my end. 

I'm using the generalized physics backend, but, for a variety of reasons, I am building the System manually instead of directly referencing a MJCF file. However, I am still using mujoco to initialize my inertias etc and was erroneously assigning the inertias to links due to ordering of the links in code vs in mujoco structures. Once I fixed the mismatch, the simulation begins to look more realistic.

However, I'm finding the simulation steps to still be slower than my use case requires due to the projected gradient solver, and so I'm looking forward to the updates you have planned there!

p.s. I've actually done away with relying on mujoco and instead use a jax implementation of trimesh's [mass_properties](https://github.com/mikedh/trimesh/blob/e62c526306766c4c597512ed7acc60c788e19e6f/trimesh/triangles.py#L199) for my (preprocessed, convex) mesh geometries

btaba, 2023-10-25:
Hi @areiner222 , thanks for the update. Have you tried messing with solver `iterations`,  `matrix_inv_iterations` and `solver_maxls`, as well as reducing unnecessary contacts and replacing meshes with primitives? If your performance is still poor, consider trying [MJX](https://github.com/google-deepmind/mujoco/tree/main/mjx) which has a better solver as # of constraints increases. However it has similar performance issues with meshes (although we pushed out a recent change that helps considerably f5d6cc5673a6ba3535204ba6a6973952cc887431).